gqrx-sdr/gqrx #362: Feature request: CTCSS and DCS codes.

DMChatterton, 2016-04-14:
It would further enhance Gqrx if identification of the above codes were to be provided for NFM signals.
Many communications receivers have this capability.

teward, 2017-08-19:
I'd love to see this as well - some things I monitor I want to ID the codes/tones on a frequency, and I wish this were a feature that was available here.

royldean, 2020-06-09:
Ok, this is a really crude method, but I've learned that if you display the "audio" interface (lower right corner), you can zoom in on the 0-250 Hz area and see a very course "spike" where the tone is located.   I don't think thee is anyway to increase the resolution to make it easier (or more precise), however.