gqrx-sdr/gqrx #365: Cannot get UDP working on Mac OS X (El Capitan)

huston007, 2016-04-18:


csete, 2016-04-18:
Hi,
Have you checked issue #363 opened just a few days ago? It sounds like the same problem

huston007, 2016-04-18:
@csete Thanks for you point.
I've just used command `nc -u -l localhost 7355 ` from this issue and got the UDP stream at last : )
So it seems to be network issue or something like that. But it doesn't work without host parameter. 

Thank you again for quick answer. My issue is closed.

csete, 2016-04-18:
Ok, well, if you don't see any data at the other end then it could be some setting in OS X. I have no idea about that, sorry. I recall other people talk about such issues some time ago, maybe it was on the mailing list, so please try to ask there.

closed 2016-04-18 by huston007

csete, 2016-04-18:
Ok, thanks for the update, and never mind my last comment.

alexf91, 2016-04-18:
I have the same problem.
Changing the UDP Host to 127.0.0.1 instead of localhost fixes this problem for me. In my case, this is caused by IPv6 (at least I think it is) and not by gqrx.

csete, 2016-04-18:
@alexf91 @huston007 Thanks for the info. I have added it to the web page about this topic at http://gqrx.dk/doc/streaming-audio-over-udp