gqrx-sdr/gqrx #369: Crash on startup

cinemast, 2016-04-25:
Hi!

First, thank you for developing gqrx, it makes it really easy also for beginners to dig into sdr.

I am currently facing problems with starting gqrx. It immediately throws an exception.
Maybe you can help me with this? 

Here is a backtrace:

```
gdb ./gqrx                  
GNU gdb (Debian 7.10-1+b1) 7.10
Copyright (C) 2015 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from ./gqrx...(no debugging symbols found)...done.
(gdb) run
Starting program: /home/cinemast/playground/gqrx/build/gqrx 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
linux; GNU C++ version 5.3.1 20160409; Boost_105800; UHD_003.009.003-0-unknown

gr-osmosdr 0.1.4 (0.1.4) gnuradio 3.7.9
built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy redpitaya 
Using Volk machine: sse4_2_64_mmx_orc
terminate called after throwing an instance of 'boost::exception_detail::clone_impl<boost::exception_detail::error_info_injector<boost::system::system_error> >'
  what():  send_to: Operation not permitted

Program received signal SIGABRT, Aborted.
0x00007ffff33c5478 in __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:55
55	../sysdeps/unix/sysv/linux/raise.c: No such file or directory.
(gdb) bt
#0  0x00007ffff33c5478 in __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:55
#1  0x00007ffff33c68fa in __GI_abort () at abort.c:89
#2  0x00007ffff3cdafcd in __gnu_cxx::__verbose_terminate_handler() () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#3  0x00007ffff3cd8f56 in ?? () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#4  0x00007ffff3cd8fa1 in std::terminate() () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#5  0x00007ffff3cd91b9 in __cxa_throw () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#6  0x00007ffff5df01de in ?? () from /usr/lib/x86_64-linux-gnu/libgnuradio-blocks.so.3.7.9
#7  0x00007ffff5df14f7 in boost::asio::detail::do_throw_error(boost::system::error_code const&, char const*) ()
   from /usr/lib/x86_64-linux-gnu/libgnuradio-blocks.so.3.7.9
#8  0x00007ffff5e126fc in unsigned long boost::asio::basic_datagram_socket<boost::asio::ip::udp, boost::asio::datagram_socket_service<boost::asio::ip::udp> >::send_to<boost::asio::mutable_buffers_1>(boost::asio::mutable_buffers_1 const&, boost::asio::ip::basic_endpoint<boost::asio::ip::udp> const&) ()
   from /usr/lib/x86_64-linux-gnu/libgnuradio-blocks.so.3.7.9
#9  0x00007ffff5e1067b in ?? () from /usr/lib/x86_64-linux-gnu/libgnuradio-blocks.so.3.7.9
#10 0x00000000004c9670 in udp_sink_f::udp_sink_f() ()
#11 0x00000000004c9cc0 in make_udp_sink_f() ()
#12 0x000000000048e4ee in receiver::receiver(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, unsigned int) ()
#13 0x000000000047a974 in MainWindow::MainWindow(QString, bool, QWidget*) ()
#14 0x000000000042736d in main ()
(gdb)
```

cinemast, 2016-04-25:
It appears that the exception gets thrown in the following line:

https://github.com/csete/gqrx/blob/master/src/interfaces/udp_sink_f.cpp#L60

csete, 2016-04-25:
Some details about OS type and version, as well as how gqrx was installed (source, binary package) could be useful.

I have never seen this error before so I probably can't help. There seems to be some permission errors while creating the UDP socket. Do you have some strict security enabled or something like that?

cinemast, 2016-04-25:
Hi!

Thank you for your quick response.

It's a Debian testing system. The package there is broken (but also outdated), so forget about that. So I cloned from your git repository and compiled it by hand. This however results in the same previously mentioned error at startup.

No specific security stuff is installed.

So I guess you are not able to reproduce it.

cinemast, 2016-04-25:
If I replace "localhost" with "127.0.0.1" it works. But seems to be a problem unrelated to you. Sorry for the noise.

closed 2016-04-25 by cinemast

csete, 2016-04-25:
Hmm, ok, thanks for the info.

cinemast, 2016-04-25:
If you have the following line in your `/etc/hosts` the application crashes as mentioned. 

`::1     localhost ip6-localhost ip6-loopback`

Just posting it here if anyone else encounters the problem.

csete, 2016-04-26:
Thanks for the info @cinemast 
Do you know if this is the default configuration in Debian? If yes, I should probably change the default in gqrx to avoid crashing.

cinemast, 2016-04-26:
Actually this setting seems to be default in Debian, however I tried to reproduce it on another machine yesterday and I couldn't. So it must the something that I screwed up locally on my machine. 

Sorry for the noise. I don't think you should change something at the moment.

steve95, 2018-12-08:
Hi, I'm experiencing the same crash on my Arch Linux where I have `::1 localhost` in /etc/hosts (a pretty standard conf, it's also mentioned in the [Arch install guide](https://wiki.archlinux.org/index.php/Installation_guide#Network_configuration)). Commenting out that line everything works fine.