gqrx-sdr/gqrx #371: Bug: Switching RTL-SDR -> UHD IO device causes crash

gnieboer, 2016-05-02:
On a Win64 MSVC 2015 build of gqrx:

> 
> 1. Run gqrx from fresh config (opens)
> 2. Select RTL-SDR device, click OK
> 3. Reopen IO selector
> 4. Select UHD device, click OK (gqrx will crash)
> 5. Restart gqrx, keep same UHD device selected, click OK (gqrx will crash)

Debugging points to a crash in the UHD dll when gqrx is setting the rx::gain

After selected the RTL-SDR, a "gains" line is added to the gqrx config file.  After the UHD device is selected, the same gains line is still present.  If that line is deleted manually, gqrx will start normally.

It appears that RTL-specific gains information is stored when they are selected, and then it is not deleted when UHD is selected.

csete, 2016-05-02:
I think this is the same as issue #125.
There is an easy workaround, namely to use different configurations for different devices. Specific configuration can be loaded using the -c command line option.

gnieboer, 2016-05-02:
It does look like the same issue.

csete, 2016-05-20:
Closing as duplicate of #125

closed 2016-05-20 by csete