gqrx-sdr/gqrx #376: Aliasing with SDRPlay RSP

Analias, 2016-05-14:
Alex,

I'm getting aliasing across all bands and demodulators with my SDRPlay RSP. It only happens in GQRX - CubicSDR and osmocom_fft seem to work fine. The aliased signals are greatly reduced if I turn off I/Q balancing, but are still present if you look close enough. Reducing the tuner gain does not make a difference.

I have included two screenshots to show you what I'm seeing.  One is gqrx at 50Mhz, the other is osmocom_fft also at 50Mhz (no aliases shown).

-Freeman

[image]
[image]

csete, 2016-05-14:
I wasn't aware that SDRPlay support actually worked. I don't have the device myself so I can't test.

In any case, are you saying that if you use the same sample rate, bandwidth and gain settings as in osmocom_fft and you have I/Q balancing off, then you get more aliasing then with osmocom_fft?

Also note that I/Q balancing only corrects I/Q phase and gain errors and can not do anything about aliasing. That must be a s setting in the hardware. Aliasing means that if you are sampling e.g. 50-52 MHz and have a signal at 52.5 MHz, then you will also see it at 51.5 MHz.

But your screenshot suggests images, which are removed by I/Q balancing. I/Q balancing does not work very well. It must be manually re-activated when the frequency is changed significantly.

csete, 2016-05-14:
I tried it with a limesdr and indeed it seems in some cases I/Q balancing has the opposite effect, so best not to use it until we figure out what's wrong. So, please make the comparison with I/Q balancing off, thanks.

csete, 2016-05-20:
Any updates here? It would be nice to confirm if there is a bug or not.

Analias, 2016-05-20:
I tried doing what you asked, and noticed I was getting odd results with osmocom_fft with varying the sampling rate.  Sometimes I would see aliases sometimes not. Frequently results would vary between identical consecutive invocations of the same command line for osmocom_fft. At this point I don't think it's a problem unique to gqrx. I could be gnuradio itself or the SDRPlay driver. I'm less likely to believe it's gnuradio, but I never see the problem when using CubicSDR which uses the same driver libraries.

Hopefully LimeSDR funds and I get a second device to test with, with only the SDRPlay my options are limited. Perhaps I should get an airspy or an RTL dongle, but I don't want to spend my limited ham radio funds on anything other than SDR devices with FPGAs that have more computational capability.

csete, 2016-05-21:
Well, eventough they use the same library, they may use it differently.

If gr-osmosdr is built with SoapySDR support, you can use the SoapySDR interface in gqrx by entering soapy=0,driver=sdrplay in the device string -- I'm not sure about the driver part though, that's just a guess.

csete, 2016-08-17:
I borrowed and SDRPlay and got it working through the SoapySDR API, i.e. using soapy=0,driver=sdrplay as device string.

With some sample rates I see many alias signals, with other rates I saw none. It should be the same with CubicSDR using the same sample rate. I will give it a try later.

pinkavaj, 2022-11-07:
@csete, @N5FPP  any update on this?

argilo, 2023-04-29:
Closing, as there has been no update and this seems unlikely to be an issue with Gqrx itself.

closed 2023-04-29 by argilo