gqrx-sdr/gqrx #377: crash when zooming out from channel

W-M-D, 2016-05-18:
Program received signal SIGSEGV, Segmentation fault.
0x0000000000548d80 in CPlotter::getScreenIntegerFFTData(int, int, float, float, long long, long long, float*, int*, int*, int*) ()

#0  0x0000000000548d80 in CPlotter::getScreenIntegerFFTData(int, int, float, float, long long, long long, float*, int*, int*, int*) ()
#1  0x000000000054e619 in CPlotter::draw() ()
#2  0x00000000004b0c7c in MainWindow::iqFftTimeout() ()
#3  0x00000000005655f7 in ?? ()
#4  0x00007ffff4887f8a in QMetaObject::activate(QObject*, int, int, void**) ()
   from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#5  0x00007ffff4894838 in QTimer::timerEvent(QTimerEvent*) () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#6  0x00007ffff4888e13 in QObject::event(QEvent*) () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#7  0x00007ffff514bffc in QApplicationPrivate::notify_helper(QObject*, QEvent*) ()
   from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#8  0x00007ffff51514b6 in QApplication::notify(QObject*, QEvent*) ()
   from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#9  0x00007ffff48595ab in QCoreApplication::notifyInternal(QObject*, QEvent*) ()
   from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#10 0x00007ffff48ae85d in QTimerInfoList::activateTimers() () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#11 0x00007ffff48aed61 in ?? () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#12 0x00007fffee6571a7 in g_main_context_dispatch () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#13 0x00007fffee657400 in ?? () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#14 0x00007fffee6574ac in g_main_context_iteration () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#15 0x00007ffff48afa3f in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) ()
   from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#16 0x00007ffff4856d6a in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) ()
   from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#17 0x00007ffff485ee0c in QCoreApplication::exec() () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#18 0x0000000000459f3c in main ()

csete, 2016-05-18:
Thanks, but could you please provide a proper report with info about what version of the application you are using and what exactly you are doing. I don't know what zooming out from channel means.

W-M-D, 2016-05-19:
linux; GNU C++ version 5.3.1 20160429; Boost_105800; UHD_003.009.004-0-unknown

I mean focusing the bandwidth to a smaller area on the waterfall. Sorry i'm not really sure how else to explain it.

W-M-D, 2016-05-19:
New FFT rate: 30 Hz
"Spectrum zoom: 1.1x"
"Spectrum zoom: 1.2x"
"Spectrum zoom: 1.4x"
"Spectrum zoom: 1.5x"
"Spectrum zoom: 1.7x"
"Spectrum zoom: 1.9x"
"Spectrum zoom: 2.1x"
"Spectrum zoom: 1.9x"
"Spectrum zoom: 1.7x"
"Spectrum zoom: 1.6x"
"Spectrum zoom: 1.4x"
"Spectrum zoom: 1.3x"
"Spectrum zoom: 1.2x"
"Spectrum zoom: 1.1x"
"Spectrum zoom: 1.0x"
"Spectrum zoom: 0.9x"
"Spectrum zoom: 0.8x"
"Spectrum zoom: 0.7x"
"Spectrum zoom: 0.7x"
"Spectrum zoom: 0.6x"
"Spectrum zoom: 0.6x"
"Spectrum zoom: 0.5x"
"Spectrum zoom: 0.5x"
"Spectrum zoom: 0.4x"
"Spectrum zoom: 0.4x"
"Spectrum zoom: 0.3x"
"Spectrum zoom: 0.3x"
"Spectrum zoom: 0.3x"
"Spectrum zoom: 0.3x"
"Spectrum zoom: 0.2x"
"Spectrum zoom: 0.2x"
Program received signal SIGSEGV, Segmentation fault.

W-M-D, 2016-05-19:
https://www.youtube.com/watch?v=MfspkslneOA&feature=youtu.be

csete, 2016-05-19:
Ok, thanks. I got it now and can reproduce it.

W-M-D, 2016-05-20:
ah ok i'll try having a go at your code.

closed 2016-05-27 by csete

pyrog, 2016-10-03:
Same problem when zooming out with the mouse wheel (mouse pointer over the frequency axis).
The waterfall width is smaller than the "window"

macOS Sierra (Mac OS X 10.12)
gqrx **2.5.3** (**2.4** on the crash report / Gqrx.app/Contents/Info.plist)

[image]

```
Process:               gqrx [25034]
Path:                  /Applications/Gqrx.app/Contents/MacOS/gqrx
Identifier:            dk.gqrx.www
Version:               2.4 (0)
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           gqrx [25034]
User ID:               501

Date/Time:             2016-10-03 19:28:13.571 +0200
OS Version:            Mac OS X 10.12 (16A323)
Report Version:        12
Anonymous UUID:        D8C2DA4D-53F0-6F41-7DF7-B7D19B7D22EF

Sleep/Wake UUID:       7687384A-91A4-47CE-9F39-3CB952AA6B1C

Time Awake Since Boot: 320000 seconds
Time Since Wake:       41000 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGBUS)
Exception Codes:       KERN_PROTECTION_FAILURE at 0x00000001320a47e8
Exception Note:        EXC_CORPSE_NOTIFY

Termination Signal:    Bus error: 10
Termination Reason:    Namespace SIGNAL, Code 0xa
Terminating Process:   exc handler [0]

VM Regions Near 0x1320a47e8:
    VM_ALLOCATE            00000001320a3000-00000001320a4000 [    4K] rw-/rwx SM=PRV  
--> VM_ALLOCATE            00000001320a4000-00000001320a5000 [    4K] r--/r-- SM=PRV  
    MALLOC_LARGE           00000001320a5000-00000001320d7000 [  200K] rw-/rwx SM=PRV  

Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   dk.gqrx.www                   	0x000000010a6ad323 CPlotter::getScreenIntegerFFTData(int, int, float, float, long long, long long, float*, int*, int*, int*) + 675
1   dk.gqrx.www                   	0x000000010a6ac34e CPlotter::draw() + 430
2   dk.gqrx.www                   	0x000000010a597ba3 MainWindow::iqFftTimeout() + 259
3   org.qt-project.QtCore         	0x000000010c1ce2e2 QMetaObject::activate(QObject*, int, int, void**) + 2994
4   org.qt-project.QtCore         	0x000000010c1c6bd0 QObject::event(QEvent*) + 48
5   org.qt-project.QtWidgets      	0x000000010b42a4bb QApplicationPrivate::notify_helper(QObject*, QEvent*) + 251
6   org.qt-project.QtWidgets      	0x000000010b42d89e QApplication::notify(QObject*, QEvent*) + 8318
7   org.qt-project.QtCore         	0x000000010c19d983 QCoreApplication::notifyInternal(QObject*, QEvent*) + 115
8   org.qt-project.QtCore         	0x000000010c1ef846 QTimerInfoList::activateTimers() + 1270
9   libqcocoa.dylib               	0x000000010ef65472 0x10ef44000 + 136306
10  com.apple.CoreFoundation      	0x00007fff75c62551 __CFRUNLOOP_IS_CALLING_OUT_TO_A_SOURCE0_PERFORM_FUNCTION__ + 17
11  com.apple.CoreFoundation      	0x00007fff75c43637 __CFRunLoopDoSources0 + 423
12  com.apple.CoreFoundation      	0x00007fff75c42bb6 __CFRunLoopRun + 934
13  com.apple.CoreFoundation      	0x00007fff75c425b4 CFRunLoopRunSpecific + 420
14  com.apple.HIToolbox           	0x00007fff751e3f6c RunCurrentEventLoopInMode + 240
15  com.apple.HIToolbox           	0x00007fff751e3ca9 ReceiveNextEventCommon + 184
16  com.apple.HIToolbox           	0x00007fff751e3bd6 _BlockUntilNextEventMatchingListInModeWithFilter + 71
17  com.apple.AppKit              	0x00007fff738da5f5 _DPSNextEvent + 1093
18  com.apple.AppKit              	0x00007fff73fea8eb -[NSApplication(NSEvent) _nextEventMatchingEventMask:untilDate:inMode:dequeue:] + 1637
19  com.apple.AppKit              	0x00007fff738cefbd -[NSApplication run] + 926
20  libqcocoa.dylib               	0x000000010ef6608d 0x10ef44000 + 139405
21  org.qt-project.QtCore         	0x000000010c19b07d QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) + 381
22  org.qt-project.QtCore         	0x000000010c19df5a QCoreApplication::exec() + 346
23  dk.gqrx.www                   	0x000000010a58bf0c main + 4556
24  libdyld.dylib                 	0x00007fff8ac82255 start + 1

```

alexf91, 2016-10-03:
It's possible that this was already fixed with #415

csete, 2016-10-03:
I have tagged version 2.6 yesterday, which means that a new binary is coming out this week.

csete, 2016-10-03:
@alexf91 

> It's possible that this was already fixed with #415 

Didn't you mean #381?

alexf91, 2016-10-03:
@csete Right, that's the one.
I thought that #381 was already included in v2.5.3.

PR #381 (2016-05-27): Fix integer overflow in plotter.cpp
Fixes #377