gqrx-sdr/gqrx #378: Xfce screensaver causes `gqrx` to stutter, even after screen is unlocked again

courtarro, 2016-05-20:
I'm not sure exactly what is happening. I was recording IQ data and the screen locker activated. When I unlocked the screen, I saw a couple `O`s appear in the console to indicate that the UHD driver had hit some buffer overruns. I then realized that `gqrx` was a bit choppy, updating the main FFT and waterfall plot less smoothly. To reproduce:

1. Use Xubuntu 14.04 (Trusty) 64-bit
2. Fire up `gqrx`
3. Start the DSP engine at 250kS/s
4. Observe that the FFT and spectrum plot are smoothly updating and that there are no `O`s appearing in the console
5. Lock the console (Xfce menu, `Lock`)
6. Unlock using your normal password
7. Observe that `gqrx` is now sluggish and `O`s are occasionally appearing in the console.

I'm using a USRP B210 over USB3 plugged straight into the computer (no hubs). When `gqrx` is laggy, the CPU usage is still quite low, so I don't think it's maxing out the CPU usage. The stuttering happens whether I'm recording IQ or not.

courtarro, 2016-05-20:
Okay, I've determined that the fault lies somewhere in the demodulation / audio playback. I disabled demod (set the `Mode` to `Demod Off`). Here's a new way to test this:

1. Fire up `gqrx`
2. Start the DSP engine at 250kS/s
3. Observe that the FFT and spectrum plot are smoothly updating and that there are no `O`s appearing in the console
4. Enable demodulation, set `Mode` to `WFM (stereo)`
5. Hear that static (or music) is playing through the speakers
6. Lock the console (Xfce menu, `Lock`)
7. Hear that the sound stops
8. Unlock using your normal password
9. Hear that the sound resumes *but* observe that `gqrx` is now sluggish and `O`s are occasionally appearing in the console
10. Change `Mode` to `Demod Off`
11. Sound stops *but* observe that the FFT is *no longer* sluggish

This test was performed with a RTL-SDR and had the same behavior as the B210.

csete, 2016-05-20:
> I've determined that the fault lies somewhere in the demodulation / audio playback

I think the problem is more likely where execution of the processing blocks is scheduled. It seems that both hardware ends (USB and audio) can put the gnuradio scheduler (or something else around there) into a weird mode that causes this behaviour.

When you change the mode (you could change it to any other mode, even the same one) the gnuradio flowgraph is reset and whatever caused it seems to go away for a while.

I have no idea what I can do about it in gqrx, except replacing the gnuradio runtime with something else, but that would practically mean a different application.

You could try rebuilding gqrx to use gr-audio instead of pulseaudio. Some people have reported that it works much better.

courtarro, 2016-05-20:
For what it's worth, if I leave the `Mode` set to `Demod Off`, the slowdown *does not* happen. I can go into and out of screensaver and the app will keep running. I can also tell that the spectrum waterfall is continuous while the screensaver was active.

How do I build using gr-audio instead of pulseaudio?

csete, 2016-05-20:
If you build using cmake there is the option LINUX_AUDIO_BACKEND, i.e. you can pass
```
-DLINUX_AUDIO_BACKEND=Pulseaudio
-DLINUX_AUDIO_BACKEND=Gr-audio

```
to the cmake step.

If you build using qmake, you have to edit gqrx.pro file and comment out line :+1: https://github.com/csete/gqrx/blob/master/gqrx.pro#L33

csete, 2016-08-11:
Hi,

Could you please check to see if you still have these problems? The above commit fixes all audio stuttering issues that I have been able to reproduce and hopefully also fixes your problems.

csete, 2016-10-14:
I can't reproduce it myself.

closed 2016-11-03 by csete