gqrx-sdr/gqrx #383: Gqrx not display line in FFT

rfrail3, 2016-06-02:
Pulling the 2.5.3 tag version from the repo and compiling without problem in a Debian 7.0 box. The ftt upper line don't appear, but the bottom waterfall yes, as normal.

I tried other releases, from this one until v2.4 but all have the same problem. Maybe is related with some library? It's a bit strange because the bottom right box with the Audio line works ok.

[image]

```
$ /usr/local/bin/gqrx 
linux; GNU C++ version 4.7.2; Boost_104900; UHD_003.007.002-0-unknown
gr-osmosdr 0.1.3 (0.1.3) gnuradio 3.7.5
built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy 
Using Volk machine: avx_64_mmx_orc
BookmarksFile is /home/user33/.config/gqrx/bookmarks.csv
gr-osmosdr 0.1.3 (0.1.3) gnuradio 3.7.5
built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy 
Using device #0 Realtek RTL2838UHIDIR SN: 00000001
Found Rafael Micro R820T tuner
```

csete, 2016-06-02:
You are not the first one to report this extremely weird issue and I have no idea what the error could be. 

Has there never been any plot in the upper range or did it suddenly disappear?

Try playing with the FFT settings and reset the configuration using gqrx -r

rfrail3, 2016-06-02:
Solved with "gqrx -r"!
Before that, I played with the FFT settings but it didn't had any difference.

Thanks,

closed 2016-06-02 by rfrail3

csete, 2016-06-02:
Ok, cool. If it suddenly disappears again and you can reproduce it, do let me know!

reox, 2016-07-26:
I had the same issue just now. I used gqrx from debian repo (v 5.4.0 20160609) and did not see the FFT. The waterfall was working.
Also `gqrx -r` helped me in this case.
I probably had a configuration file of an older version of gqrx on the system and I suspect some interference there. 
I tested it on another machine with no previously installed gqrx and the fft showed up just like that.

I'm not really sure if this is just a configuration file problem but in my case it looks like it.

delta-alpha, 2018-02-09:
I had the same issue. I clicked on the FFT tab and moved the averaging slider all the way to the left. I'm not sure what changed to cause this to stop, to begin with, this wasn't a tab that I had ever accessed before.

csete, 2018-02-09:
Thanks @delta-alpha , this gives me a clue that the bug could be in the averaging code.

sam210723, 2018-11-20:
Same issue in 2.11.3 under Windows 10. FFT springs back to life after moving the Averaging slider any amount.