gqrx-sdr/gqrx #384: Release device when DSP processing is off

yuvadm, 2016-06-11:
As gqrx is currently implemented, an RTLSDR device is acquired upon launch and only released when the gqrx process terminates. This means that any attempt to acquire the device during this time in any other process will fail. This is not very elegant behavior.

Is there any limitation why not to release the device when there's no DSP processing taking place?

csete, 2016-06-11:
Gqrx is just a GUI and doesn't have any direct access to the devices. You probably want to take a look at the start/stop functions in gr-osmosdr.

alexf91, 2016-06-11:
At least for the BladeRF, the start/stop functions don't release the device.
It only gets released when the Top-Block is destroyed.

yuvadm, 2016-06-11:
@csete is it possible to use gr-osmosdr in any other way that resolves this issue?

alexf91, 2016-06-11:
The start and stop functions are not exposed by the public interface of gr-osmosdr.
I don't think there is a way other than destroying the source when DSP is disabled and creating a new one when it is enabled again.

yuvadm, 2016-06-11:
@alexf91 is there any significant performance penalty (or otherwise) in doing that?

csete, 2016-06-11:
That's what happens when you open the I/O configurator and press the OK button. It destroys the device and creates a new one. The problem with that - aside from the performance penalty - is that one is not supposed to do that with GNU Radio objects. Basically, we a re just lucky that it works.

As I said, the right place to add device open / close is in the start / stop functions of the backend.

alexf91, 2016-06-11:
Performance is probably not an issue here, but the correct way to achieve the use case you mentioned would be the implementation of raw-IQ streaming.

yuvadm, 2016-06-11:
@csete's comment makes sense to me. If no other workaround exists in this case, I'll take this upstream to gr-osmosdr.

csete, 2016-10-01:
Ok, I'm closing this issue since there is nothing do in gqrx.

closed 2016-10-01 by csete