gqrx-sdr/gqrx #385: More shortcuts or user defined keybindings

LongnoseRob, 2016-06-15:
The bookmark list can be shown oder hidden with Ctrl-B. 
If you do not  have the main toolbar shown there is no shortcut to add the current frequency to the bookmarks. Or I did not find the shortcut yet.

How about adding Shift-Ctrl-B for creating a bookmark?

I remember with some graphics tool kits it was possible to add user defined shortcuts, is this also possible with QT?

Thank you for the great software!

csete, 2016-06-15:
The same functionality should be available regardless of whether the toolbar is visible or not. I will take a look at it when I get some time, though this has a relatively simple solution so a patch is welcome if anybody decides to take a stab at it.

LongnoseRob, 2016-08-14:
I had a small look into this issue again and found an easy way to add Ctrl+Shift+B to the Add Bookmark entry in the mainwindo.ui file.

[shortcut-diff.txt](https://github.com/csete/gqrx/files/416921/shortcut-diff.txt)

csete, 2016-08-14:
Thanks for the patch.

I have added the shortcut, but note that even this shortcut does not work if the toolbar is hidden. As a workaround, I have also added the "Add bookmark" action to the menu bar and this makes the shortcut work -- as long as the menu bar is visible.

So, I guess we can close this issue now.

LongnoseRob, 2016-08-15:
No problem,

I think I when i tested the new shortcut, I did not pay attention if the main menu bar was shown or not

Yes, this issue can be closed

closed 2016-08-15 by csete