gqrx-sdr/gqrx #386: missing icons on Suse Leap 42.1

codekraft-studio, 2016-06-20:
I just installed **gqrx** to test an antenna just bought it and it works perfectly, the only problem, though not dramatic, is that are missing GUI icons.

I installed from here: [software.opensuse](https://software.opensuse.org/package/gqrx?search_term=gqrx)

When I launch I got this: 

![no-icons](https://lh3.googleusercontent.com/greRucmshx92J-foFjzTPDiFb5vaj5QhB014K9NpxdkHRv8mPiEShF3i625WbWlQCZmI-kD3GAP_nR1tM3IpD5Rzb4TR6Dokv_8F53xnDHIWI_B7rJYtlt9R6ima0I8u2bp9NeL220MLremEklwIe8KF1mQvGKrHWowJDkBIX62FND5ySVN03E3HzUCWI8A374xBcE3Rptbn8qFiU--fGRyqH-vx91TXlU5Dc1zDiWAVUwP2J0Zh7DIkuVHeNsfRjcnz7TYOK5hZ9usRSCzpLiKXH_MK2To_9wPMDOiJ8deNB_1Efs8kTgcCxHg1qiH6vEzYjDqf4kCSwRagU02l-sydSZ2NFTf4Vp4PR3Wwv99ZHFOxrNO8zH1u7O4kKLeRQUoEJRfZ77Qc6wdpTB3jqmJ0Bg-ErMdgXzoSi4S4KkypKwpUdE19g5pQH2asq02rufddnloCi-TVdyAe6y1vEVmVDf6xF5X9f4RRm1VnH971Q2Xj0QSNv_3qzlnGJ7QrY2PYtYZ32U_zvUCZJ_RKgo3rlHja-WyR5I_vhOfWfYlLubc1JKQSv9nh4NuV5IjW7v4Uynb_bGYjgwyautjCqm_SD_yT9x4=w955-h130-no)

by the way thanks for this great tool.

csete, 2016-06-20:
Gqrx has the QtSvg package as runtime dependency to load icons in SVG format. I don't know the exact name in Suse, but look for something like libqt5-svg. Or maybe version 4... Once you find the library please report it to the packager so they can update the package dependencies.

codekraft-studio, 2016-06-27:
I just red your reply and downloaded the package: `libQt5Svg5` on Suse and everything works correctly, many thanks!

closed 2016-06-27 by codekraft-studio