gqrx-sdr/gqrx #387: Make ctrl+scroll zoom in/out FFT

d33tah, 2016-06-24:
Sometimes I'd like to be able to zoom in in gqrx and switching to "FFT settings" tab to change that isn't too conivenient. Could you bind zoom in/out behavior to ctrl+scroll?

csete, 2016-06-24:
You can use the mouse wheel on the frequency and dB axis to zoom. You can also drag the axes to pan. The widget in the FFT tab were only added recently in case mouse zooming works bad with some touch pads.

d33tah, 2016-06-28:
Ah! Nice, that solves the problem for me. Though from UX perspective, adding the ctrl+scroll might still be pretty intuitive.

csete, 2016-06-28:
I assume with "UX perspective" you mean something like a standard image viewing application or map  where ctrl+scroll will zoom in and out. Ok, fine... I understand that. However, in SDR zooming both X and Y axes at the same time makes little sense, at least in my opinion.

So, what should ctrl+scroll zoom, X or Y?

csete, 2016-06-28:
Perhaps we can use ctrl+scroll for zoom in frequency and shift+scroll to zoom in dB?

d33tah, 2016-06-29:
@csete: yup, that makes sense. shift+scroll for "db range", ctrl+scroll for "zoom" from FFT settings tab.

Analias, 2016-07-02:
FYI - Ctrl-Scroll (I assume "Scroll" refers to the scroll wheel on the mouse) on the Mac is used to zoom the entire Mac desktop. I don't see a way of overriding the default behavior for the Mac.

Analias, 2016-07-02:
I did a little bit more research - it is possible to disable the UI Ctrl-Scroll on the Mac in the Accessibility system preferences under the Zoom settings. You can change the keyboard modifier from Ctrl to Option or Command key, but that's not very useful in this use case.  Disabling the UI scrolling re-enables the gqrx Ctrl-Scroll.

csete, 2016-07-02:
Thanks for the info @Analias . It looks like these desktop zoom shortcuts are disabled by default.