gqrx-sdr/gqrx #389: Incorrect bandwidth with IQ sources

Cat-Ion, 2016-07-05:
I'm using a pipe to get raw IQ data into gqrx. The bandwidth is fixed at 96 kHz, even if I specify a higher rate:

    file=/tmp/iq,freq=443e6,rate=5e6,repeat=false,throttle=false

I assume this is because gqrx doesn't check the bandwidth of file sources and keeps the previously selected bandwidth, which is 96 kHz if the receiver is freshly constructed.

csete, 2016-07-05:
Did you try to see the messages printed in the terminal with a debug build? It should show what sample rate it uses and possibly also why.

Cat-Ion, 2016-07-05:
When starting, [it outputs this stuff](https://gist.github.com/Cat-Ion/7c5fd1468c47be02b61c5deef1e035eb) - Not sure what the relevant bits are, but I can see "updateHWFrequencyRange failed fetching new hardware frequency range" in line 89 and just before that it complains that the actual bandwidth is 0 Hz.

csete, 2016-07-05:
Yes, it's probably not good that the source block returns 0 Hz bandwidth but I don't remember the details.

Cat-Ion, 2016-07-05:
The source block never calls the function to set the input rate. I'm going to try setting it right after osmosdr::source::make() has been called in receiver::set_input_device().

Cat-Ion, 2016-07-05:
Seems to work, and without interfering with the operation of regular devices, as far as I can tell.

Cat-Ion, 2016-07-06:
Just noticed, even with this patch there seems to be some weirdness. If I select 5 Msps in gqrx, I get a decent spectrum over the full 5 MHz. If I set that up with IQ piping, the spectrum drops off sharply after ±1 MHz, by -50 dB when you reach 2 MHz. Is there some filter that I might have forgotten to set?

Cat-Ion, 2016-07-07:
It's only if the sampling rate starts differently: If you set gqrx's sampling rate to 5 Msps but at first only pipe it in at 100 ksps, then move it up to more than that, the bandwidth is weird. Works fine otherwise.

csete, 2016-07-08:
Ok, sorry I can't provide any useful feedback but I am very busy these days and don't have any energy to look into this issue.

Now I merged your pull request before I read these comments; should I revert it?

Cat-Ion, 2016-07-08:
I don't think so - as long as the sampling rates match up it seems to work fine.

argilo, 2020-11-28:
I'll close this off since it seems a fix was already merged.

closed 2020-11-28 by argilo