gqrx-sdr/gqrx #393: Feature idea: Waterfall scaling, changing colors when changing squelch level, ...

Cat-Ion, 2016-07-10:
Keeping some of the old FFT data around for the waterfall would have a bunch of advantages. It would let us zoom around even for old data, and changing the min/max dB values could be applied to old data as well.

The obvious problem might be memory usage. Maximum would be 4 MiB per line for an FFT of size 2^20. For a fullscreen window on a 1080p monitor this might use some 3 GiB of RAM, if the waterfall display uses 2/3 of the screen. Could halve that by using e.g. a 16-bit int to store dB levels instead of full floats. On the other hand, PCs that do FFTs on a million samples at once might just be beefy enough to spare those 3 GiB anyway.

Each line would have some metadata associated with it, probably min/max frequency, a timestamp, and the FFT size. A ringbuffer sized by the number of lines in the waterfall should be appropriate for storing that, plus pointers to the FFT data, and not incur much overhead when checking for and deleting old data.

Performance shouldn't change, as the data would only have to be accessed when scaling the waterfall in some way.

I know @csete is too busy, but I wouldn't mind working on this when I get some time if I'm not the only one interested in this :)

csete, 2016-07-22:
I have a few ideas of my own, which would require double buffering and I think what you propose also requires it.

Currently, the waterfall has 8-bit resolution, which I believe is sufficient. Most common SDRs supported by gqrx have a dynamic range < 120 dB, so an 8-bit palette will still give better than 0.5 dB resolution.

Cat-Ion, 2016-07-23:
Yeah, but the dynamic range of gqrx is larger, going from 0 dB down to at most -300 dB, so that's about 1.2 dB resolution. If you use a small 10 dB range (which are admittedly rather grainy even without an intermediate step) you'll be down to just 8 steps. Might be worth just testing out, or leaving it configurable (at least with a #define). We might be able to use 10-bit samples or so, but would have to check the overhead of all those bitshifts.

csete, 2016-07-23:
Ok, you lost me here, but I'm pretty sure that gqrx does not have 300 dB dynamic range.

Cat-Ion, 2016-07-23:
You can set the reference level to anywhere from 0 to -100 dB, and the dB range can be 10 to 200 dB.

Cat-Ion, 2016-07-23:
oh, what we might be able to do is scan each FFT block and store the min/max and rescale to 0-255, so we'd have

    struct FftSample {
        qint64 freq_min, freq_max;
        qint64 timestamp;
        qint64 size;
        float db_min, db_max;
        qint8 *data;
    }

Cat-Ion, 2016-08-22:
It's about time I got started on this. I'm tending towards making the buffer its own class instead of creating a bunch of new members in the waterfall class, but it doesn't really seem to fit under any of the folders in the current structure. What do you think?

Cat-Ion, 2016-12-15:
My current ideas for the size restrictions are:

1. You have some height and a maximum size. Just restrict each single FFT to maxsize/height and resample if necessary
2. If an FFT increases the size above what's allowed, remove the oldest FFT(s) until the size is okay again.
3. If an FFT increases the size above what's allowed, halve the size of the largest FFT until the size is okay again.

If more than one of these sounds desirable, it shouldn't be a problem to implement all of them and toggle between them at runtime. Personally, I'm leaning towards the third option.

Cat-Ion, 2016-12-17:
Another thing I should do is redrawing only what's needed - when the waterfall is just being scrolled on the frequency axis, we can move the picture and redraw the sides. Changing the dynamic range or zooming in would still require a redraw.

csete, 2016-12-18:
Hi @Cat-Ion 

There is one thing that bothers me about this feature request here, namely that it has a very broad and not well defined scope. For instance, what do the "..." in the headline cover? When can we say that we are finished with this?

I would prefer to have more items with smaller and well defined scope. The work you have started to do in #461 seems better limited to waterfall caching, although I still miss a clear and concise description of the desired functionality, which will make it difficult for me to review.

Can we limit the scope of this item to e.g. "Waterfall scaling and caching" with a proper description, and create new issues for other features and ideas?

Cat-Ion, 2016-12-19:
the '...' was meant to be 'all the other things that are related to changing the axes on the waterfall', so pretty much what I've implemented. The feature itself (rescaling and the needed caching) is complete, the other things I'm mentioning are definitely things that would warrant their own issue - I just didn't want to spam the tracker with it while I'm still the only one working on it :) To reflect the limited scope we could change the title to something like "Caching of old FFTs to allow rescaling waterfall axes". I'd then, if you don't mind, start opening issues for the other things I've talked about, i.e. memory limiting strategies and redrawing only parts of the waterfall for scrolling.

csete, 2016-12-20:
> To reflect the limited scope we could change the title to something like "Caching of old FFTs to allow rescaling waterfall axes". I'd then, if you don't mind, start opening issues for the other [...]

That would be great. I don't mind having many small issues open, as long as it is clear when they are done. Even for larger items I like to have a clear scope defined. By the way, Github allows using check boxes that can be used for tracking progress of larger issues for example:

- [x] Task 1
- [ ] Task 2
- [x] Task 3

csete, 2016-12-23:
I think we completely misunderstand each other. I meant to open new issues for things that could be considered as separate tasks from this one, in order to limit the scope of this task. But a sub-task is still a part of this task, so this task is still big and now it is spread out over 3 issues, which IMHO is even more complicated.

Anyway, perhaps you should just implement what you want to implement, then submit a pull request along with a description of what you have implemented. But please provide a clear and concise description of what has been implemented and why/how it is useful.

csete, 2016-12-23:
... and I wouldn't mind if we got rid of #468 and #469 or decoupled them from this one.

Cat-Ion, 2016-12-23:
Oh. Sorry about that!

I'm not sure this task can be split up more, other than simple removing #468 and #469 from it - it's functional without them, and they can be considered separate features, unless you'd consider #468 critical.

I'm going to be writing a better description of what's been changed for the existing pull request.