gqrx-sdr/gqrx #394: gqrx devel crashes

kukabu, 2016-07-12:
Hello

I builds last gqrx and it's crash.
```
$ /usr/local/bin/gqrx -c 23123.conf
linux; GNU C++ version 5.1.1 20150618 (Red Hat 5.1.1-4); Boost_105800; UHD_003.008.002-0-unknown

gr-osmosdr v0.1.x (0.1.4git) gnuradio 3.7.8.1
built-in source types: file fcd rtl rtl_tcp uhd hackrf rfspace 
Using Volk machine: sse4_a_64_orc
BookmarksFile is /home/mike/.config/gqrx/bookmarks.csv
gr-osmosdr v0.1.x (0.1.4git) gnuradio 3.7.8.1
built-in source types: file fcd rtl rtl_tcp uhd hackrf rfspace 
The RTL TCP server reports a R820T tuner with 29 RF and 0 IF gains.
terminate called after throwing an instance of 'boost::exception_detail::clone_impl<boost::exception_detail::error_info_injector<boost::math::rounding_error> >'
  what():  Error in function boost::math::round<float>(float): Value inf can not be represented in the target integer type.
Aborted (core dumped)
$ cat .config/gqrx/23123.conf 
[General]
crashed=true

[input]
device="rtl_tcp=rtl:23123"
sample_rate=240000
```
on rtl server:
```
client accepted!
set gain mode 1
set agc mode 0
set direct sampling 0
Disabled direct sampling mode
set offset tuning 0
set sample rate 0
Invalid sample rate: 0 Hz
comm recv bye
Signal caught, exiting!
worker cond timeout
Signal caught, exiting!
all threads dead..
listening...
```

kukabu, 2016-07-12:
I revert https://github.com/csete/gqrx/commit/451554eec0adbc4d146c474b1d6a2743dc147835 and gqrx works again.

Cat-Ion, 2016-07-12:
Sorry for breaking this.
This patch should fix it - it does on my system. Can you confirm that?

kukabu, 2016-07-12:
Thanks. The patch fixes bug.

csete, 2016-07-12:
Please keep testing so we can find all bugs :-)

PR #395 (2016-07-12): Do not automatically set samplerate if input reports 0
Fixes #394.
closed 2016-07-12 by csete