gqrx-sdr/gqrx #399: bookmarks on audio spectrum

kukabu, 2016-08-04:
Hello

Sometimes I see bookmarks on audio spectrum. Why?!
[image]

csete, 2016-08-04:
Hi @kukabu 

Thanks for the report. We can certainly consider this to be a bug and I think I can fix it quite easily.

You see the bookmarks on the audio spectrum because it uses the same code as the main spectrum and we didn't think about disabling bookmarks in the audio spectrum. So, bookmarks at these very low frequencies show up there.

Good that you found this out because processing the bookmarks also uses CPU and there is no reason for that here.

csete, 2016-08-06:
It should be fixed now.

closed 2016-08-06 by csete