gqrx-sdr/gqrx #401: Feature Request: Keyboard Shortcuts

jonathanendersby, 2016-08-14:
I find myself instinctively wanting to use the keyboard to tune the frequency and adjust the squelch. I think this would be non-intrusive and add a lot of value.

My suggestions would be:
* Left and Right Arrows: Decrease and Increase Frequency by 2000hz
* Up and Down: Adjust SQL by 10.
* Shift + Left and Right: Decrease and Increase Frequency by 500hz (Fine Tuning)
* Shift + Up and Down: Adjust SQL by 1 (Fine Tuning)
* Spacebar: Start/Stop DSP (similar to most audio programs)
* Cmd+Left and Right: Pan left and right by 2000hz
* Cmd+Shift+Left and Right: Pan left and right by 500hz (Fine Tuning)

csete, 2016-08-14:
Hi,

I have been looking into adding more shortcuts to allow using gqrx without a mouse. I will take your suggestions into account though I can't promise that it will be exactly like that.

Is there anything special about 500 and 2000 Hz tuning steps? Personally I would prefer different tuning steps according to mode.

Note that you can already use the arrows to adjust frequency but it requires the mouse to hover the widget, so it is not very useful.

DSP start stop has CTRL+D as shortcut. I don't think we can use spacebar as shortcut without breaking assistive technologies.

csete, 2016-08-14:
By the way, what do you mean with pan left and right? In SDR terms we can change the hardware frequency (the center with the blue line) and the filter offset (between red and blue line). I assume you mean one of these?

jonathanendersby, 2016-08-14:
Hi,

Thank you for your reply. The 500 and 2000 Hz steps were picked at random. I'm busy implementing them into https://github.com/marmelo/gqrx-remote (which is a solution, but not a great one because the app needs focus to work).

With regards to panning: I'm referring to moving the spectrum left and right. When zoomed in slightly (eg 15x) and adjusting the channel filter offset, I am able to tune "off the edge" of the spectrum. This results in a spectrum window like the following: (notice the red line on the left edge).
[image]

I can pan to the currently demodulated frequency by dragging the spectrum's frequency axis. What I'm requesting is that CMD+Left or Right pans similarly. (I'm not sure I'm using the correct words here...? Is the red line called the channel filter offset? Is there a way to have the spectrum display "follow" the red line so that it doesn't go off the side of the spectrum?)

Thanks for the CMD-D tip, don't know how I missed that!

csete, 2016-08-14:
Ok, I understand now what you mean with panning now. I didn't think about zoom and pan sounds like a good term for this. It definitely needs improvement; ideally, it would pan automatically as the channel (red line) moves close to the edge.

jonathanendersby, 2016-08-14:
Thank you! 
ps. I just pushed https://github.com/marmelo/gqrx-remote/pull/10 adding the arrow key functionality.