gqrx-sdr/gqrx #402: Gqrx 2.5.3-2 + rtl_tcp garbled demodulation and high CPU

kadin2048, 2016-08-19:
I found what appears to be a problem with Gqrx and rtl_tcp, using the precompiled Gqrx 2.5.3-2 binaries for Mac OS X.  For test purposes, I ran both Gqrx and rtl_tcp on the same machine to eliminate network issues.

When Gqrx is connected directly to the SDR hardware ("rtl=0"), everything works well.  CPU usage is about 190-210% as measured by top, when demodulating WFM-Stereo. ([Screenshot](http://i.imgur.com/wl4fEVH.png))  I also made sure the hardware was working correctly using `rtl_test` (no samples per million lost after 5 mins of running after warmup).

Then, I tried using rtl_tcp; specifically the version bundled inside Gqrx 2.5.3-2 (located at `/Applications/Gqrx.app/Contents/MacOS/rtl_tcp`). I've tried invoking rtl_tcp with various parameters; the best performance I could get was by doing: `./rtl_tcp -g 8 -s 2560000 -n 256 -f 88500000`

It starts fine:
```
Found 1 device(s):
  0:  Generic, RTL2832U, SN: [redacted]

Using device 0: Generic RTL2832U
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
Tuned to 88500000 Hz.
Tuner gain set to 8.000000 dB.
listening...
Use the device argument 'rtl_tcp=127.0.0.1:1234' in OsmoSDR (gr-osmosdr) source
to receive samples in GRC and control rtl_tcp parameters (frequency, gain, ...).
```

I then connected Gqrx to the rtl_tcp daemon with device string `rtl_tcp=127.0.0.1:1234` (which, incidentally, requires opening Gqrx once and force-quitting it, then reopening it a second time, in order to get it to present the "Gqrx has detected problems with the current configuration..." message and let me change parameters.  Otherwise it hangs, presumably because it tries to connect to rtl=0 and rtl_tcp already has control of it.  This is a minor issue in comparison, though). 

Immediately on connecting -- before even pressing the on/off button in Gqrx -- there are a lot of `ll+, now XX` messages in the rtl_tcp window, where XX ranges from 1 to 256 (because I set the buffer to 256 earlier, this makes sense).  I'm not sure if this is normal buffer-filling activity or indicative of a problem.

When I actually turn Gqrx on, I get a reasonable-looking waterfall display, but demodulated audio sounds terrible, *and* Gqrx starts consuming all available CPU in the machine.  All cores will be immediately at 95+%, net Gqrx usage is 700+%, cooling fans going full bore.  (The usage seems to all be on the Gqrx side, rtl_tcp only has something like 2% CPU.)  The waterfall display will also get "pixellated" during demodulation, presumably because of lack of CPU for the UI.  With no demodulation enabled, Gqrx CPU is around 288%.  (When idle, Gqrx is <1%.)

I can provide audio samples of both the actual speaker output, and also what Gqrx records using the record-to-disk feature.  The speaker audio is garbled and sounds "slow motion" while the on-disk audio sounds OK, but "skips" every few seconds and contains only a partial recording.

Periodically, the rtl_tcp window will show the `ll+, now XX` messages, from 1 to 256 again, but I can't discern any connection between what it does and the output in Gqrx.

The other thing I've noticed is that the frequency and gain controls in Gqrx seem to take a very long time to do anything (at least a minute, sometimes more). Moving the gain or frequency causes an immediate message to show up in the rtl_tcp window, but it's not reflected in Gqrx's output until much later.  I don't know if this is related.

I am happy to perform other tests or troubleshooting.  I also have access to a Linux machine running Gqrx 2.3.1 on Debian Jessie which _appears_ to do the same thing, but I haven't done quite as much research there.

Test equipment:
- Mac Pro 2 x 2.8 GHz quad-core (8x cores), 32GB RAM, MacOS 10.10.5 (14F1713)
- RTL2832U (generic)

csete, 2016-08-19:
Thanks for the report.

I have tagged this as a bug and will be happy to take a look at it when I get time; however I suspect the error is outside of gqrx. Gqrx does not do any hardware interfacing itself; we use the gr-osmosdr library for that and the error could be there.

In any case I would recommend reducing the sample rate below 2.4 Msps.

By the way, you can create different configurations for different devices and launch gqrx from the terminal using a specific configuration, see "gqrx -h" so you avoud conflicts with device access. Let me know if you need more info about this - it's very useful for debugging.

k9wkj, 2016-08-19:
rtl_tcp is a very odd (and i believe horribly broken) thing
i cant get it to run on a 8core cpu locally with out stuttering it self to
a screeching stop
using any client that will connect to it
the only time i have gotten it to work is running on a Pi II over a wired
connection inside my network and that ran fine
remember that rtl_tcp uses ENORMOUS amounts of bandwidth
there was a fellow who wrote a transport/control server for rtl_sdr
https://github.com/texane/librtlsdr-rpc.git  this makes the dongle appear
as a local device that works fine for everything that will connect to it
but there again bandwidth *hog*

kadin2048, 2016-08-19:
I don't think that network bandwidth is the issue given that both the rtl_tcp server and the client (Gqrx) are running on the same machine. I did think that might have been the problem, the first time I tried it (over WiFi, and then GigE) but the issue seems to be a CPU-bound problem on the receive side, whether it's in Gqrx proper or, as csete suggests, in the gr-osmosdr library.

I'd be happy to submit the bug report upstream to gr-osmosdr, though I'd imagine that it would be good if we could conclusively point to the library being the thing that's choking and consuming all the CPU, rather than Gqrx.  Doing a process sample during normal (direct USB) operation and then again during rtl_tcp operation would be easy enough, and ought to show what thread is absorbing all the CPU.  Would that be useful?

k9wkj, 2016-08-20:
ohh wasnt imp;ing its is a network issue
as if it wont run locally on a 8core beast but will run on a Arm cpu over
100Mb ethernet
its a rtl_tcp issue
and we are not the only ones to have the trouble
this isnt a GQRX issue its a rtl_tcp issue
like i said try https://github.com/texane/librtlsdr-rpc.git   it works

csete, 2016-08-21:
As mentioned in [this thread on Reddit](https://www.reddit.com/r/RTLSDR/comments/4ygu66/is_anyone_using_rtl_tcp_and_gqrx_successfully/d6nug49?st=is4d4gic&sh=d1062d31), the gr-osmosdr source block seems to work well inside GNU Radio. So the error is probably in gqrx or the way gqrx uses gr-osmosdsr.

csete, 2017-03-19:
Possibly a duplicate of issue #99.
We will provide a workaround in the binary packages for 2.7 and hopefully that will fix this issue. If not feel free to reopen or file a new issue.

closed 2017-03-19 by csete

CleversonSA, 2018-06-02:
Hello,
I was having this topic problem by days with my Raspberry Pi 3 (acting as a rtl_tcp server) and GQRX 2.6 (client). Tried some tests with grosmocom librtlsdr in a i3 4GB RAM laptop (with Debian 8) and even on a Windows 7 port, cable and Wifi. I had no luck.

So, reading this topic I tried the "texane" librtlsdr implementation and rtl-tcp worked fine on my RPi 3, without major issues.

Regards.