gqrx-sdr/gqrx #404: 'qInfo’ was not declared in this scope

janvgils, 2016-08-27:
Trying to build the latest code on Ubuntu 14.04 LTS

Running into the following problem when building the last code (27-08-2016 - 18:09)
If I read this right, then there is a function and or declare missing in dockrxopt.cpp.

It this enough information to solve this issue, or do I have add extra info on the build system?

[ 78%] Building CXX object src/CMakeFiles/gqrx.dir/qtgui/dockrxopt.cpp.o
git/gqrx/src/qtgui/dockrxopt.cpp: In member function ‘float DockRxOpt::currentMaxdev()’:
git/gqrx/src/qtgui/dockrxopt.cpp:272:11: error: ‘qInfo’ was not declared in this scope
     qInfo() << __FILE__ << __FUNCTION__ << "FIXME";
           ^
git/gqrx/src/qtgui/dockrxopt.cpp: In member function ‘void DockRxOpt::getFilterPreset(int, int, int*, int*) const’:
git/gqrx/src/qtgui/dockrxopt.cpp:292:15: error: ‘qInfo’ was not declared in this scope
         qInfo() << __func__ << ": Invalid mode:" << mode;
               ^
git/gqrx/src/qtgui/dockrxopt.cpp:297:15: error: ‘qInfo’ was not declared in this scope
         qInfo() << __func__ << ": Invalid preset:" << preset;
               ^
make[2]: *** [src/CMakeFiles/gqrx.dir/qtgui/dockrxopt.cpp.o] Error 1
make[1]: *** [src/CMakeFiles/gqrx.dir/all] Error 2
make: *** [all] Error 2

Regards, Jan van Gils

csete, 2016-08-28:
Ups... I forgot to check if this was available in the Qt that ships with Ubuntu 14.04. I will fix it right away. Thanks.

Alex

csete, 2016-08-28:
Ok, you can try again :-)

closed 2016-08-28 by csete

janvgils, 2016-08-28:
Thanks for the update, that fixed the build failure.

commit 3556cbf7 (2016-08-28): Don't use qInfo() since it's not avaialble in Qt 5.2.

Fixes #404.