gqrx-sdr/gqrx #405: Gqrx NOAA APT reception issue

lesquilter, 2016-09-06:
For the past few months I've been using Gqrx successfully to receive NOAA APT signals. However, I believe as of 2.5.3, the audio output from Gqrx now renders in WxToImg as offset blocks, rather than as a clear image. The issue _could_be with either SoX or WxToImg, but Gqrx is the only thing which has changed recently. I've tested this multiple times over the past 4 days.
Attached are some sample images, one of previous working state, one of current state. 

[image]

[image]

My apologies in advance if this is not a Gqrx issue, but all the data I can find seems to indicate that it is.

csete, 2016-09-06:
Hi @lesquilter 

Thanks for the report. I have been super busy I didn't have any time to play with this feature for a long time, so it could very well be that there is something broken in gqrx. Note that version 2.5.3 has been  out for quite a while now so perhaps you can clarify whether it was during the upgrade to 2.5.3 or the recent update in the Ubuntu packages last week that broke it for you.

I'm afraid I will not be able to fix this before version 2.6 is released, but I promise to take a look at it shortly after. In fact, I have a new demodulator optimized for APT reception in the pipeline.

lesquilter, 2016-09-21:
I'm going to close this ticket. After trying a bunch of different things, as a last resort I removed (apt-get purge) and reinstalled gqrx. Subsequently it appears to be working perfectly. Not sure how that fixed it but it has.

closed 2016-09-21 by lesquilter