gqrx-sdr/gqrx #406: GQRX and SoapySDRServer fail to connect

thehose, 2016-09-08:
I'm attempting to connect to a remote rtl_sdr dongle running on a SopySDRServer.  It works well when using osmocom_fft with these device strings:

"soapy=0,driver=remote,remote=sdr"
"soapy=0,driver=remote,remote=sdr,remote:driver=rtlsdr"

When I try with gqrx (2.5.3) I does not connect and the server shows no connection attempted. Attached is the conf file used.
[Soapy_RTL_SDR.txt](https://github.com/csete/gqrx/files/462789/Soapy_RTL_SDR.txt)

PS: I have the SoapySDRServer installed and working. I would be glad to run any test's you may need in order to debug this issue.

Hose

csete, 2016-09-08:
Will try to take a look at it for the 2.7 release. Any more info and data is of course welcome :-)

thehose, 2016-09-08:
Absolutely, I’ll keep poking around and if anything comes from that I’ll let you know. I’m expecting a LimeSDR at the end of this year (?) and then I’ll switch to that device and put the dongle to eternal rest.

csete, 2017-12-10:
Hi,

I just tried using SoapySDRServer with the LimeSDR and RTL-SDR and it works. The device string I used was:

> soapy=0,driver=remote,remote:driver=lime

and 

> soapy=0,driver=remote,remote:driver=rtlsdr

respectively.

So, it appears to be working with the latest software.

closed 2017-12-10 by csete

GianisTsol, 2019-11-10:
How can i specify the address of the soapyRemoteServer?

szpajder, 2019-11-10:
`soapy=0,driver=remote,remote=<ip_address_or_hostname>,remote:driver=<remote_driver_name>`