gqrx-sdr/gqrx #407: Qmake is failing this time

c0debreaker, 2016-09-12:
make was fine few days ago until today after doing a `git pull`

```
g++ -c -m64 -pipe -O2 -Wall -W -D_REENTRANT -DQT_WEBKIT -DQT_NO_DEBUG -DQT_NO_DEBUG_OUTPUT -DVERSION="\"v2.5.3-106-g2d5f\"" -DWITH_PULSEAUDIO -D_REENTRANT -DQT_NO_DEBUG -DQT_GUI_LIB -DQT_NETWORK_LIB -DQT_CORE_LIB -DQT_SHARED -I/usr/share/qt4/mkspecs/linux-g++-64 -I../../gqrx -I/usr/include/qt4/QtCore -I/usr/include/qt4/QtNetwork -I/usr/include/qt4/QtGui -I/usr/include/qt4 -I../src -I. -I. -I../../gqrx -I. -o nbrx.o ../src/receivers/nbrx.cpp
../src/receivers/nbrx.cpp: In constructor ‘nbrx::nbrx(float, float)’:
../src/receivers/nbrx.cpp:53:14: error: no match for ‘operator=’ (operand types are ‘resampler_ff_sptr {aka boost::shared_ptr<resampler_ff>}’ and ‘int’)
     audio_rr = 0;
              ^
../src/receivers/nbrx.cpp:53:14: note: candidates are:
In file included from /usr/include/boost/shared_ptr.hpp:17:0,
                 from /usr/include/gnuradio/sptr_magic.h:26,
                 from /usr/include/gnuradio/basic_block.h:27,
                 from /usr/include/gnuradio/block.h:27,
                 from /usr/include/gnuradio/sync_block.h:27,
                 from /usr/include/gnuradio/analog/simple_squelch_cc.h:27,
                 from ../src/receivers/nbrx.h:26,
                 from ../src/receivers/nbrx.cpp:25:
/usr/include/boost/smart_ptr/shared_ptr.hpp:505:18: note: boost::shared_ptr<T>& boost::shared_ptr<T>::operator=(const boost::shared_ptr<T>&) [with T = resampler_ff]
     shared_ptr & operator=( shared_ptr const & r ) BOOST_NOEXCEPT
                  ^
/usr/include/boost/smart_ptr/shared_ptr.hpp:505:18: note:   no known conversion for argument 1 from ‘int’ to ‘const boost::shared_ptr<resampler_ff>&’
/usr/include/boost/smart_ptr/shared_ptr.hpp:514:18: note: template<class Y> boost::shared_ptr<T>& boost::shared_ptr<T>::operator=(const boost::shared_ptr<Y>&) [with Y = Y; T = resampler_ff]
     shared_ptr & operator=(shared_ptr<Y> const & r) BOOST_NOEXCEPT
                  ^
/usr/include/boost/smart_ptr/shared_ptr.hpp:514:18: note:   template argument deduction/substitution failed:
../src/receivers/nbrx.cpp:53:14: note:   mismatched types ‘const boost::shared_ptr<X>’ and ‘int’
     audio_rr = 0;
              ^
In file included from /usr/include/boost/shared_ptr.hpp:17:0,
                 from /usr/include/gnuradio/sptr_magic.h:26,
                 from /usr/include/gnuradio/basic_block.h:27,
                 from /usr/include/gnuradio/block.h:27,
                 from /usr/include/gnuradio/sync_block.h:27,
                 from /usr/include/gnuradio/analog/simple_squelch_cc.h:27,
                 from ../src/receivers/nbrx.h:26,
                 from ../src/receivers/nbrx.cpp:25:
/usr/include/boost/smart_ptr/shared_ptr.hpp:525:18: note: template<class Y> boost::shared_ptr<T>& boost::shared_ptr<T>::operator=(std::auto_ptr<_Tp1>&) [with Y = Y; T = resampler_ff]
     shared_ptr & operator=( std::auto_ptr<Y> & r )
                  ^
/usr/include/boost/smart_ptr/shared_ptr.hpp:525:18: note:   template argument deduction/substitution failed:
../src/receivers/nbrx.cpp:53:14: note:   mismatched types ‘std::auto_ptr<Y>’ and ‘int’
     audio_rr = 0;
              ^
In file included from /usr/include/boost/shared_ptr.hpp:17:0,
                 from /usr/include/gnuradio/sptr_magic.h:26,
                 from /usr/include/gnuradio/basic_block.h:27,
                 from /usr/include/gnuradio/block.h:27,
                 from /usr/include/gnuradio/sync_block.h:27,
                 from /usr/include/gnuradio/analog/simple_squelch_cc.h:27,
                 from ../src/receivers/nbrx.h:26,
                 from ../src/receivers/nbrx.cpp:25:
/usr/include/boost/smart_ptr/shared_ptr.hpp:543:77: note: template<class Ap> typename boost::detail::sp_enable_if_auto_ptr<Ap, boost::shared_ptr<T>&>::type boost::shared_ptr<T>::operator=(Ap) [with Ap = Ap; T = resampler_ff]
     typename boost::detail::sp_enable_if_auto_ptr< Ap, shared_ptr & >::type operator=( Ap r )
                                                                             ^
/usr/include/boost/smart_ptr/shared_ptr.hpp:543:77: note:   template argument deduction/substitution failed:
/usr/include/boost/smart_ptr/shared_ptr.hpp: In substitution of ‘template<class Ap> typename boost::detail::sp_enable_if_auto_ptr<Ap, boost::shared_ptr<T>&>::type boost::shared_ptr<T>::operator=(Ap) [with Ap = int]’:
../src/receivers/nbrx.cpp:53:14:   required from here
/usr/include/boost/smart_ptr/shared_ptr.hpp:543:77: error: no type named ‘type’ in ‘struct boost::detail::sp_enable_if_auto_ptr<int, boost::shared_ptr<resampler_ff>&>’
Makefile:672: recipe for target 'nbrx.o' failed
make: *** [nbrx.o] Error 1
```

csete, 2016-09-12:
Thanks for the heads up, but could you please provide info about you Qt and gcc version?

c0debreaker, 2016-09-12:
gcc version 4.9.2 (Debian 4.9.2-10) and qt4

csete, 2016-09-12:
Hmm, it's building fine with gcc 4.9.2 and Qt5... I will see if I can fix it later, but it is becoming increasingly difficult to maintain compatibility with Qt4.

Is there any particular reason why you stay with Qt4?

csete, 2016-09-12:
Maybe it is not Qt4 anyway... Can you try to replace

`audio_rr = 0;`

with

`audio_rr = NULL;`

and see if it makes a difference.

c0debreaker, 2016-09-12:
Oh, I can upgrade to QT5. When I installed it, I didn't pay attention to the QT I was installing

c0debreaker, 2016-09-12:
here is the result of make. it still failed
```
../src/receivers/nbrx.cpp: In constructor ‘nbrx::nbrx(float, float)’:
../src/receivers/nbrx.cpp:54:14: error: no match for ‘operator=’ (operand types are ‘resampler_ff_sptr {aka boost::shared_ptr<resampler_ff>}’ and ‘long int’)
     audio_rr = NULL;
              ^
../src/receivers/nbrx.cpp:54:14: note: candidates are:
In file included from /usr/include/boost/shared_ptr.hpp:17:0,
                 from /usr/include/gnuradio/sptr_magic.h:26,
                 from /usr/include/gnuradio/basic_block.h:27,
                 from /usr/include/gnuradio/block.h:27,
                 from /usr/include/gnuradio/sync_block.h:27,
                 from /usr/include/gnuradio/analog/simple_squelch_cc.h:27,
                 from ../src/receivers/nbrx.h:26,
                 from ../src/receivers/nbrx.cpp:25:
/usr/include/boost/smart_ptr/shared_ptr.hpp:505:18: note: boost::shared_ptr<T>& boost::shared_ptr<T>::operator=(const boost::shared_ptr<T>&) [with T = resampler_ff]
     shared_ptr & operator=( shared_ptr const & r ) BOOST_NOEXCEPT
                  ^
/
```

csete, 2016-09-12:
Ok, if it happens with Qt5 also it could be boost related. I guess I just have to try understand what the heck it means.

csete, 2016-09-12:
Can you try:

`audio_rr = nullptr;`

and:

`audio_rr.reset();`

csete, 2016-09-12:
Ok, I tried to build with Qt4 and get the same error. Changing it to audio_rr.reset(); seems to fix it for me so I am going to make the change. Please confirm if it works for you too, thanks!

c0debreaker, 2016-09-12:
It worked! :)

c0debreaker, 2016-09-12:
I want to learn how to code in this language as well as code for my rtl-sdr. I want to write a command line version that I can wrap with Node.JS. Any suggestions on where I should start?

csete, 2016-09-12:
The high level SDR interface in gqrx is in src/applications/gqrx/receiver.h and src/applications/gqrx/receiver.cpp - that's what I would wrap in a command line version.

c0debreaker, 2016-09-12:
Thanks. I'm going to take a look at it. I'll close this issue now. Thanks for fixing the issue. :)

closed 2016-09-12 by c0debreaker