gqrx-sdr/gqrx #412: Segfault while starting gqrx

W-M-D, 2016-09-24:
Program received signal SIGSEGV, Segmentation fault.
strchr () at ../sysdeps/x86_64/multiarch/../strchr.S:32
32	../sysdeps/x86_64/multiarch/../strchr.S: No such file or directory.

csete, 2016-09-24:
Sorry, but I can't do anything about it unless you write a proper bug report.

W-M-D, 2016-09-24:
What information would you like ?

W-M-D, 2016-09-24:
After building/installing from source it worked fine.

csete, 2016-10-09:
Minimum info usually includes:

- Operating system, e.g. linux or Mac
- Distribution, e.g. Debian 32 or 64 bit
- Installed from repository, source, or something else
- What SDR hardware is being used

Since you now have it working, do you agree to close the issue?

closed 2016-10-16 by W-M-D