gqrx-sdr/gqrx #424: CloudIQ stops streaming after ~5 minutes of inactivity

csete, 2016-10-01:
If gqrx is left running for about 5 minutes with the CloudIQ, the radio will suddenly stop streaming. Interestingly, the LEDs on the radio suggest that it is still streaming to the host. No messages in the serial console.

Probably the same issue as #429, i.e. the CloudIQ expects either some control items or keepalive packets at regular intervals.

csete, 2016-10-09:
There is now a workaround for this in the binary distributions (PPA, Mac OS, RPI). The source patch is [available here](https://github.com/csete/gr-osmosdr-gqrx/commit/a533be4c3ecc7013ad61694935deba8383cc726c) until upstream is updated.

argilo, 2020-11-28:
Closing, since this was fixed in gr-osmosdr: https://git.osmocom.org/gr-osmosdr/commit/?id=6ea6c19028f8a93c3a942856def5afcf894cba9d

closed 2020-11-28 by argilo