gqrx-sdr/gqrx #425: recipe for target 'dockfft.o' failed

mayorbobster, 2016-10-02:
Hello.  When attempting to build gqrx, I get:

In file included from ../src/qtgui/dockfft.cpp:28:0:
./ui_dockfft.h: In member function ‘void Ui_DockFft::setupUi(QDockWidget*)’:
./ui_dockfft.h:108:21: error: ‘class QScrollArea’ has no member named ‘setSizeAdjustPolicy’
         scrollArea->setSizeAdjustPolicy(QAbstractScrollArea::AdjustToContentsOnFirstShow);
                     ^
./ui_dockfft.h:108:41: error: ‘AdjustToContentsOnFirstShow’ is not a member of ‘QAbstractScrollArea’
         scrollArea->setSizeAdjustPolicy(QAbstractScrollArea::AdjustToContentsOnFirstShow);
                                         ^
Makefile:656: recipe for target 'dockfft.o' failed
make: *** [dockfft.o] Error 1

My build script looks a bit like:
#!/bin/bash
CORES=`cat /proc/cpuinfo | grep "cpu cores" | tail -1 | sed 's,^.*: ,,'`
# assume hyperthreading
CORES=`expr $CORES \*  2`
if [ ! -d gqrx.git ]
then
git clone https://github.com/csete/gqrx.git gqrx.git 
cd gqrx.git
else
cd gqrx.git
git pull
fi
# git clone https://github.com/csete/gqrx.git gqrx.git
export CFLAGS="-fPIC"
BOOST_ROOT=/data/boost/`ls /data/boost | grep z | grep boost | sort -V | tail -1 | sed 's,[.].*,,'`
rm -rf build
mkdir build
cd build
rm -rf ./*
qmake ..
make -j $CORES
make install
chown me.users `which gqrx`

alexf91, 2016-10-02:
QScrollArea has no method called setSizeAdjustPolicy in Qt 4.8, but it has one in Qt 5.

As mentioned in the README, versions prior to Qt 5 are not officially supported and might not work anymore.

csete, 2016-10-02:
Indeed, it is becoming difficult to maintain Qt 4 compatibility. I have now updated both the readme and the gqrx.pro file to require Qt 5. I believe Qt 5 is now available on all platforms where gqrx runs so this shouldn't be a problem.

@mayorbobster Which distribution are you using?

mayorbobster, 2016-10-03:
Hello.  Thank you for the reply.

I suspected it might be a library/compatibility issue.  I'm running a 
(recently updated) Linux mint (18).  I haven't changed any Qt libraries, 
and a (partial) view of the installed Qt shows:

ii  libqt5gui5:amd64 5.5.1+dfsg-16ubuntu7.1                     
amd64        Qt 5 GUI module
ii  libqt5help5:amd64 5.5.1-3build1                              
amd64        Qt 5 help module
ii  libqt5network5:amd64 5.5.1+dfsg-16ubuntu7.1                     
amd64        Qt 5 network module
ii  libqt5opengl5:amd64 5.5.1+dfsg-16ubuntu7.1                     
amd64        Qt 5 OpenGL module
ii  libqt5positioning5:amd64 5.5.1-3ubuntu1                             
amd64        Qt Positioning module
ii  libqt5printsupport5:amd64 5.5.1+dfsg-16ubuntu7.1                     
amd64        Qt 5 print support module
ii  libqt5qml5:amd64 5.5.1-2ubuntu6                             
amd64        Qt 5 QML module
ii  libqt5quick5:amd64 5.5.1-2ubuntu6                             
amd64        Qt 5 Quick library
ii  libqt5quickwidgets5:amd64 5.5.1-2ubuntu6                             
amd64        Qt 5 Quick Widgets library
ii  libqt5script5:amd64 5.5.1+dfsg-2build1                         
amd64        Qt 5 script module
ii  libqt5sensors5:amd64 5.5.1-1ubuntu2                             
amd64        Qt Sensors module
ii  libqt5sql5:amd64 5.5.1+dfsg-16ubuntu7.1                     
amd64        Qt 5 SQL module
ii  libqt5sql5-sqlite:amd64 5.5.1+dfsg-16ubuntu7.1                     
amd64        Qt 5 SQLite 3 database driver
ii  libqt5svg5:amd64 5.5.1-2build1                              
amd64        Qt 5 SVG module
ii  libqt5test5:amd64 5.5.1+dfsg-16ubuntu7.1                     
amd64        Qt 5 test module
ii  libqt5webkit5:amd64 5.5.1+dfsg-2ubuntu1                        
amd64        Web content engine library for Qt
ii  libqt5widgets5:amd64 5.5.1+dfsg-16ubuntu7.1                     
amd64        Qt 5 widgets module
ii  libqt5x11extras5:amd64 5.5.1-3build1                              
amd64        Qt 5 X11 extras
ii  libqt5xml5:amd64 5.5.1+dfsg-16ubuntu7.1                     
amd64        Qt 5 XML module

I will check my libraries for setSizeAdjustPolicy.  It seems it was 
introduced in Qt 5.2, but... when I run qmake --version it

tells me : Qmake version 2.01a  Using Qt version 4.8.7 in 
/usr/lib/x86_64-linux-gnu.   On further research, I can either add a

switch -qt=qt5 or set an environment variable QT_SELECT=qt5 which will 
force it to build using the newer version of QT.

so qmake -qt=qt5 --version  gives me

Qmake version 3.0
Using Qt version 5.5.1 in /usr/lib/x86_64-linux-gnu

...and now my script looks like:

#!/bin/bash
CORES=`cat /proc/cpuinfo | grep "cpu cores" | tail -1 | sed 's,^.*: ,,'`
# assume hyperthreading
CORES=`expr $CORES \*  2`
if [ ! -d gqrx.git ]
then
git clone https://github.com/csete/gqrx.git gqrx.git
cd gqrx.git
else
cd gqrx.git
git pull
fi
# git clone https://github.com/csete/gqrx.git gqrx.git
export CFLAGS="-fPIC"
BOOST_ROOT=/data/boost/`ls /data/boost | grep z | grep boost | sort -V | 
tail -1 | sed 's,[.].*,,'`
rm -rf build
mkdir build
cd build
rm -rf ./*
qmake -qt=qt5 ..
make -j $CORES
make install
chown me.users `which gqrx`

...and it is running very nicely.  The issue may only pop up for people 
who upgrade systems rather that installing from scratch. Dmesg -l | Grep 
qt gives me

new versions of qt, but a lot of qt4 libraries too.

Thanks again

csete, 2016-10-03:
It is normal that a desktop system will have both Qt4 and Qt5 installed as many legacy applications still require Qt4. You can normally also enforce Qt5 for your build by using qmake-qt5 or by installing the qt5-default package.

Anyhow, you are now up an running again so I hope we can close this issue, right?

mayorbobster, 2016-10-04:
Yes, close the issue.

closed 2016-10-04 by csete

mayorbobster, 2016-10-08:
Hi.  I thought I replied to you about this already.  Yes, close the 
issue.  GQRX has been running fine since I found the switch (and/or 
environment variable).

Thanks again,

Bob