gqrx-sdr/gqrx #426: Zoom cursor is not refreshed when zooming with the mouse

pyrog, 2016-10-03:
In this example, the zoom is around **1** but the cursor is always at **34x**
[image]

macOS Sierra
gqrx 2.5.3

csete, 2016-10-03:
Yeah, the two things are out of sync due to the way they were done historically, but it's such a minor issue that I doubt I will ever have time to fix it.

alexf91, 2016-10-03:
Looks like I do have time ;-)
#428

csete, 2016-10-03:
> Looks like I do have time ;-)
> #428

That's great :-)
I'm busy releasing v2.6 this week, but I will take a look at it afterwards.

csete, 2016-10-06:
Pull request has been merged. Will be available in version 2.7

closed 2016-10-06 by csete