gqrx-sdr/gqrx #429: Remote control: only the last connection working

kukabu, 2016-10-04:
Hello

Run multiple the remote control connections to the gqrx. All connection are allowed and accepted but the gqrx works only with the last connection.

alexf91, 2016-10-04:
Indeed.
There are a couple of ways to deal with this issue:
* Allow multiple connections
* Refuse new connections
* Close old connections

#430 provides a fix for this by closing old connections.

kukabu, 2016-10-04:
Thanks.
As workaround it's well but I think allow multiple connections is better way.

kukabu, 2016-10-04:
hm... refuse new connections as workaround is better, a new connection can be an accident or a mistake and in this case, we run the risk of breaking something.

alexf91, 2016-10-04:
Refusing new connections might cause some trouble if the old connection is lost without a proper close/shutdown of the socket.
AFAIK there's no reliable, portable way to detect a lost connection, except for sending something.
Many protocols use heartbeat messages to work around this problem, but Gqrx only responds to commands.

kukabu, 2016-10-04:
I think SO_KEEPALIVE with custom TCP_KEEPCNT/TCP_KEEPIDLE/TCP_KEEPINTVL will be enough it our case.

The GQRX is interactive application and user can be restarted server when found troubles (warning message may be useful hints for this).

csete, 2016-10-07:
Gqrx can only manage a single connection and the correct behavior is to ignore new connections as long as there is an active one. I was sure this is how it worked, but I may remember it incorrectly, or perhaps things have changed in time.

If a connection becoming inactive is a concern, we can add a timeout that checks that a command has been received during the last period (e.g. 1 minute), if not, it terminates the connection.

DjSteker, 2024-01-10:
In setting you give access to the remote IPEndPoint to the application, that is, to the IP of the application client, it works for me in the format ::ffff:127.0.0.1  "is LocalHosr"
::ffff:192.168.1.1   "is WAN "
::ffff:192.168.1.2   "is Server GQRX localhost"
::ffff:192.168.1.3  "is Cliente TCP" 
¿?

rag2, 2024-01-16:
DJ,

the address you've quoted is an IPv6 compatibility representation of an IPv4 address, such as one might use on a network using mixed  IPv6 **and** IPv4 addresses, as noted at https://en.wikipedia.org/wiki/IPv6_address#Representation. There are all sorts of reasons why this might be necessary (I'm not sure if GQRX only supports IPv4 addresses, but there are many other reasons.)

HTH, 73,

Robin, G8DQX
On 10/01/2024 21:58, DjSteker wrote:
> In setting you give access to the remote IPEndPoint to the application, that is, to the IP of the application client, it works for me in the format ::ffff:127.0.0.1 "is LocalHosr"
> ::ffff:192.168.1.1 "is WAN "
> ::ffff:192.168.1.2 "is Server GQRX localhost"
> ::ffff:192.168.1.3 "is Cliente TCP"
> ¿?

PR #1417 (2025-03-24): Allow multiple connections in remote control
Fixes #429.

Modified the remote control server so multiple clients can remote control the program at the same time.
closed 2025-03-24 by argilo