gqrx-sdr/gqrx #431: Raspberry Pi code

yo2ldk, 2016-10-05:
Hi,
this is not an issue, more an request
 Alexandru, can you make an image for Raspberry Pi and that have included GPIO for control the software; like external buttons for VFO, BAND, MODE ?
 Here is a good source for that: [http://abyz.co.uk/rpi/pigpio/](url)

   Thank you, 73 !

LongnoseRob, 2016-10-05:
Interesting request,
However I think it should be better not to done in gqrx directly.

I would rather create an interface wrapper from gpio to the remote control interface of gqrx.

Just my 2cents

yo2ldk, 2016-10-05:
that it was my first tough :) 
 if it is more easy to make link between soft and gpio control 
please consider..

 73 !

csete, 2016-10-07:
There is another feature request #401 talking about keyboard shortcuts. If we have keyboard shortcuts for all basic functions we can extend that to physical controls using a Teensy board.

When connected to a computer a Teensy board can emulate keyboard and mouse events. So, if we connect our buttons and knobs to the Teensy they can be translated to keyboard and mouse events using simple (?) Arduino code. This would provide a solution that could work on all platforms regardless of GPIO availability.

What do you guys think about this idea?

yo2ldk, 2016-10-07:
Sure, this can be done easily, but to be honest, why we spend money for another module, when Pi have it all on GPIO sure, but a little more work to integrate in . Best part is maybe you can win something:
 [https://apache-labs.com/al-products/1043/10Mhz-High-Stability-100-PPB-TCXO-for-Daughtercard.html](url)  and this not have posibility to change that more SDR devices like GQRX...

csete, 2016-10-07:
The concept with Teensy has already been proven last year by devnulling, see [https://www.youtube.com/watch?v=1YYz0kW67oU](https://www.youtube.com/watch?v=1YYz0kW67oU) or his project here on Github: https://github.com/devnulling/etch-a-sdr

It can become a generic solution and not just a Pi specific solution. Anyway, I'm pretty sure there is a way to translate GPIO signals into keyboard events or remote control commands.

closed 2017-08-26 by csete