gqrx-sdr/gqrx #443: IPv6

ED6E0F17, 2016-10-27:
GQRX needs to open a UDP port on localhost, or it will not start.

GNUradio expects localhost to be INET (127.0.0.1)
Some systems resolve localhost to INET6 (::1)
https://github.com/gnuradio/gnuradio/commit/689f507e9d5cbbac76742f5515ec925067e92bfb

Changing  "localhost" to "127.0.0.1" in the GQRX source code allows it to start.

I assume that this is my own fault for enabling some IPv6 features when upgrading my kernels

LongnoseRob, 2016-10-27:
I think the problem is that it does not open a dual stack port.. 
but it could also be a config issue on the system side.

Had similar issue under Archlinux with v6 enabled by default..

kukabu, 2016-10-27:
I agree with @LongnoseRob also I often use [::1], this is shorter :).

LongnoseRob, 2016-10-27:
I use ::ffff:127.0.0.1 that made atleast gqrx,gpredict and direwolf working..

pinkavaj, 2022-11-07:
I cannot reproduce this, is this still valid, what are the steps to reproduce it?

argilo, 2022-11-07:
I suspect this was resolved by #1088. I'll close this off.

closed 2022-11-07 by argilo