gqrx-sdr/gqrx #445: Frequency widget digits set to 0 on retune

ly2ss, 2016-10-31:
(git version)
 Changing frequency 10ths or 100ths MHz sets lesser digits to 0 
I don't know if this is intended feature or some bug.
AFAIR earlier versions behaved different - all lesser digits remained unchanged.
 F.e.: Let's tune to the middle of ham band 14.05MHz and change MHz to 21MHz -  I was used that gqrx tuned to 21.05 but now it's tuned to 21MHz.
 IMHO (as a ham) leaving them unchanged is preferred way,

csete, 2016-10-31:
Hi @ly2ss 

This was an intended change, so not a bug, sorry.

 Can you try and see if it works out for you? If not, perhaps I can make it a configuration option.

ly2ss, 2016-10-31:
ok, I'll try to get used to it.

csete, 2016-10-31:
Ok, but please reopen if you think it's too annoying. I'm sure we can find a way to make it configurable, even if a hidden config value.

devnulling, 2016-10-31:
IMO it'd be nice to have this as a configurable option with the old method. The new way is usable, but the old way was useful as well. It throws off the causal scanning by adjusting more than what the selected step change is.

csete, 2016-10-31:
Okay, lets reopen it.

FYI, it is an easy mod if you want to have it locally until we make the change, see commit https://github.com/csete/gqrx/commit/5f7c0177f4d8084a1700284443f3bfd05cc92678 .

ly2ss, 2016-11-01:
Thanks for mod! Returned back to old behavior. I give my vote for config option too.

csete, 2017-01-09:
@ly2ss @devnulling 
Can you please check if a manual config record introduced with commit e7f3c94 is sufficient? Thanks.

ly2ss, 2017-01-17:
Confirming - it's working and sufficient. Thank you.

csete, 2017-03-19:
Now it is even configurable from the GUI.

closed 2017-03-19 by csete

ly2ss, 2017-03-19:
perfect! Thanks!