gqrx-sdr/gqrx #446: where i can find source gqrx raspberry pi3

stackprogramer, 2016-11-01:
hi  i want to source gqrx for raspberry pi, is this source on github is possible use for raspberrypi3?
i want to built gqrx on raspberry from soucre.
best regards stackprogramer

csete, 2016-11-01:
Hi @stackprogramer 

Gqrx on the RPI3 uses the same source code, i.e. the one here. But in order to make it run well on RPI2 and RPI3, I had to:

1. Build gcc compiler for RPI3 that can generate proper code. The one included in raspbian is not good enough.
2. Build all dependencies from source
3. Build gqrx from source

stackprogramer, 2016-11-01:
thank you very much

csete, 2016-11-03:
Do you need any more info or can we close this?

stackprogramer, 2016-11-05:
thanks very much,i didn't test compile gqrx on raspberry,but it is hard to compile with processor raspberry 3,any way it will be take long times..............
any way thanks

closed 2017-01-07 by csete