gqrx-sdr/gqrx #447: GQRX remote control not accepting anything other than "::ffff:127.0.0.1"

Brett99, 2016-11-02:
I am trying to add a trusted computer to the list of hosts, and it is rejecting all connections. I can only telnet to localhost:7356 when "::ffff:127.0.0.1" is in the list, but "localhost" or "127.0.0.1" don't work on their own, and neither to addresses of other computers.

Brett99, 2016-11-02:
I was able to connect to it with another computer by using ::ffff:192.168.0.8
Maybe put something in the documentation telling people they need to use the ::ffff: ? The picture on the website just has normal ipv4 addresses in the box.

csete, 2016-11-03:
Ideally, it should work with both IPV4 and IPV6 adresses.

martin3dsc, 2018-07-04:
I have same problem. perhaps I mis a setting. 
No telnet from host to remote gqrx system.
What do I wrong??

localhost      = 10.72.192.10 (gqrx system) raspberi-pi
remote hosts= 10.72.192.50    ubuntu 18-04

~/.config/gqrx/default.conf
[remote_control]
allowed_hosts=10.72.192.50, 127.0.0.1, 10.72.192.0/24, ::ffff:127.0.0.1, ::1
enabled=true

  Telnet on remote host         
$ telnet 10.72.192.10 7356
Trying 10.72.192.10...
Connected to 10.72.192.10.
Escape character is '^]'.
Connection closed by foreign host.

   Telnet on localhost
$ telnet localhost 7356
Trying ::1...
Connected to localhost.
Escape character is '^]'.
f
28523000
q
Connection closed by foreign host.

alexf91, 2018-07-04:
Please post the terminal output of Gqrx. It should say something like "Remote connection attempt from <address> (not in allowed list)".

Also note that specifying multiple hosts like 10.72.192.0/24 doesn't work.

martin3dsc, 2018-07-04:
I edit the remote setting
I added ::ffff to the IP addresses the remote host succeeded in a telnet session
Problem solved!

now this is the remote controle sessing in the default.conf
[remote_control]
allowed_hosts=::ffff:127.0.0.1, ::ffff:10.72.192.50, ::ffff:10.72.192.10

argilo, 2022-01-25:
#1088 will allow ordinary IPv4 addresses to be used in the remote control settings dialog, which should make things easier.

closed 2022-01-25 by argilo