gqrx-sdr/gqrx #449: UDP functionality not working

dborg89, 2016-11-03:
Hello,
I have tried using both Ubuntu 16.04 LTS and Linux Mate 18 with the GQRX PPA repo's (v2.6)
The issue I have is that when I try and start UDP and then netcat or netstat 7355 is dead.

I have changed IP on both installs to 127.0.0.1 and their network address (192.168.x.x) and neither work.

Thanks

csete, 2016-11-03:
Have you tried IPV6 address?

csete, 2016-11-03:
I mean something like ::ffff:127.0.0.1

dborg89, 2016-11-04:
Hi and thanks for the fast reply. IPv6 is disabled on my setup.

csete, 2016-11-04:
In that case I have no idea. I have tested it again, although on Xubuntu 14.04, and it works as expected. I may try Xubuntu 16.04 later but I am quite busy at the moment.

Just to be sure... The UDP stream only sends audio samples, so gqrx must be actively receiving with a mode that produces audio for any UDP traffic to exist.

csete, 2016-11-04:
I have now also tested on Xubuntu 16.04 and here too it works as it should.

dborg89, 2016-11-05:
I will try on Xubuntu 16.04 shortly.
Just for the edification of this topic I will elaborate on the hardware.
The OS is installed inside a VMWare (Workstation) guest. 
Parent hardware is an i5 with 16GB ram (del Optiplex 9020).

When you ran the install did you use your own PPA repo (IE: used the guide: http://gqrx.dk/download/install-ubuntu)?

Thanks

dborg89, 2016-11-05:
Just an update, I installed a Xubutnu VM and ran GQRX via the PPA and this is the netstat output:
dborg@ubuntu:~$ netstat -aut
Active Internet connections (servers and established)
Proto Recv-Q Send-Q Local Address           Foreign Address         State
tcp        0      0 ubuntu:domain           *:*                     LISTEN
tcp        0      0 *:ssh                   *:*                     LISTEN
tcp        0      0 localhost:ipp           *:*                     LISTEN
tcp        0      0 localhost:6010          *:*                     LISTEN
tcp        0    168 192.168.0.20:ssh        192.168.0.19:54564      ESTABLISHED
tcp        0      0 192.168.0.20:ssh        192.168.0.19:54565      ESTABLISHED
tcp6       0      0 [::]:ssh                [::]:*                  LISTEN
tcp6       0      0 ip6-localhost:ipp       [::]:*                  LISTEN
tcp6       0      0 ip6-localhost:6010      [::]:*                  LISTEN
udp        0      0 ubuntu:domain           *:*
udp        0      0 *:bootpc                *:*
udp        0      0 *:54453                 *:*
udp        0      0 *:mdns                  *:*
udp        0      0 *:ipp                   *:*
udp6       0      0 [::]:mdns               [::]:*
udp6       0      0 [::]:43559              [::]:*
dborg@ubuntu:~$

csete, 2016-11-05:
Yes, I used our setup guide and PPA exactly as written.

Try sending UDP packets in the same direction using nc and see if they get through.

dborg89, 2016-11-05:
Hi csete,
I used sendip to test UDP port opening on 7355 and listened via nc.
Using root to run sendip worked fine, using a standard user account to send failed.
I started GQRX as root this time (just to be safe and ensure permissions aren't stopping it), however no change :(

root@ubuntu:~# sendip -p ipv4 -p udp -ud 7355 -d "Testing 7355 on UDP" -v 127.0.0.1
Added 26 options
Initializing module ipv4
Initializing module udp
Finalizing module udp
Finalizing module ipv4
Final packet data:
45 00 00 2F   E../
69 BF 00 00   i...
FF 11 53 FC   ..S.
7F 00 00 01   ....
7F 00 00 01   ....
00 00 1C BB   ....
00 1B AC 55   ...U
54 65 73 74   Test
69 6E 67 20   ing
37 33 35 35   7355
20 6F 6E 20    on
55 44 50   UDP
Sent 47 bytes to 127.0.0.1
Freeing module ipv4
Freeing module udp

dborg@ubuntu:~$ nc -l -u 127.0.0.1 7355
Testing 7355 on UDP

dborg89, 2016-11-05:
Just an update, I tried compiling from source using qmake and there is not change in the situation :/
I can't figure out what I might be missing?

Cheers

csete, 2016-11-05:
I have no idea, sorry. But I don't think it's a bug in gqrx.

PS: You have never confirmed that you are actually running the application and have audio in the speakers. It's important because UDP packets are only sent when there is audio.

dborg89, 2016-11-05:
I have tested the audio out via the soundcard output and all is working fine on that front.
Is there a debugging option that I can enable to see if the UDP trigger gets activated upon pressing the button?

Thanks

csete, 2016-11-06:
There is no debug option in gqrx, but not much is happening in gqrx itself. You press the button, it calls a function, which calls a function i gnu radio.

dborg89, 2016-11-06:
Okay, so with that being said the issue may be with GNU Radio then.
I will install GNU Radio from source and might try to trace the application's call to see if it's actually launching GNU Radio.

Thanks for the information, I will update in the next 24-48 hours.

closed 2017-08-26 by csete