gqrx-sdr/gqrx #450: Unstable audio pitch on CW / warbled SSB sound / RTL-SDR (R82xx chip. PLL not locked errors appear)

mike6000, 2016-11-04:
I am trying to run gqrx on my RPI w/ Raspian OS. 

I experience unstable audio pitch on CW  and warbled SSB sound 
The donge contains an RTL-SDR (R82xx chip). I have PLL not locked errors. 

My RTL-SDR donge works fine on Windows under HDSDR: crystal clear and stable audio. I have the same problem with two different RTL-SDR donges (of the same type) on my RPI, but both works perfectly OK on my Windows box. This leads me to believe that there must be a software issue. 
I have even tried to use different power supplies on my RPI, just to check if there could be a power issue since the device is powered from the RPI USB port of course. No change in warbled sound.

Is this a bug or config error from my side?

csete, 2016-11-04:
This is a typical indication of CPU overload. How did you install gqrx?

mike6000, 2016-11-05:
It is the audio pitch that changes. Its not like a buffer overrun sound. Sounds like an "unstable vfo" to describe it with analog terminology.

I installed by running:
$ sudo apt-get update
$ sudo apt-get upgrade
   (reboot if necessary)
$ sudo apt-get install libqt5gui5 libqt5core5a libqt5network5
$ sudo apt-get install libqt5widgets5 libqt5svg5 libportaudio2
$ wget https://github.com/csete/gqrx/releases/download/v2.6/gqrx-2.6-rpi3-1.tar.xz
$ tar xvf gqrx-2.6-rpi3-1.tar.xz
$ cd gqrx-2.6-rpi3-1
$ ./setup_gqrx.sh

csete, 2016-11-05:
Ok, just wanted to make sure you were using our packages according to the instructions. As you can see on the page, the setup has been verified to work up to 1.44 Msps	sample rate.

You shouldn't get the error message when you start gqrx. It suggests that the configuration file didn't get saved during last exit.

rtl_test does not produce any sound, so even if it runs without error it could still be reading samples with unstable audio pitch. Try rtl_fm. Also check if there are any relevant messages in syslog using the dmesg command.

900fy, 2016-12-05:
I have the same unstable audio pitch on SSB problem on Raspberry PI 3 and RTL-SDR. I've been trying different sample rates but it didn't help. CPU load is about 40%-60% but audio is distorted. I use the latest Raspbian version. 

I also have GQRX 2.6 installed on my laptop with Ubuntu 16.04 and audio is crystal clean with the same RTL-SDR dongle.

csete, 2016-12-05:
Thanks for the info @900fy 

I will try to investigate as soon as I can, but I am just moving into a new house that requires all my time for the rest of the year, so it may take some time. Until then, any more info is welcome.

mike6000, 2016-12-30:
Tested now again on a different RPI 3 computer with a fresh Raspian image with latest updates and upgrades, two different RTL-SDR dongles (same type). Used also two different HF frequency converters in front of the RTL-SDR dongles. Both experience the warbled CW sound. Listened to a commercial AM broadcaster while  in USB mode. The carrier beat tone warbles. Both RTL-SDRs and both frequency converters work nice on the PC running Win 7 with HDSDR so it must be something with the sw.

csete, 2016-12-30:
Thanks for the update @mike6000.

I adjusted the title to reflect that this is an issue with running gqrx on the Raspberry Pi.

mike6000, 2017-01-12:
I have after further investigation found that the issue I had seems to be caused by powering the RTL-SDR and frequency converter with the inbuilt USB port in the RPI. When I switched to an externally powered USB-hub, the wobble / unstable audio problem disappeared.

closed 2017-01-12 by mike6000

csete, 2017-01-12:
@mike6000 Thanks for following up on it!

900fy, 2017-01-15:
@mike6000 thanks for this info, today I plugged my up-converter to different power source instead of USB port on RPI and it also solved my problem! Thanks one more time!