gqrx-sdr/gqrx #452: Ubuntu 16.10 crash.

michelinok, 2016-11-17:
Hi!
I'm not able to run gqrx from Ubuntu 16.10 fresh install.
Tryed from sources and from package via pt-get install gqrx-sdr.
It always crash with segafult when i try to run it

[  345.079830] gqrx[4235]: segfault at 1 ip 00007f7d9f00c188 sp 00007fff052da390 error 4 in libgnuradio-blocks.so.3.7.10[7f7d9ee91000+2bc000]

Any idea?

csete, 2016-11-17:
Hi @michelinok 

Please see this thread:
https://groups.google.com/forum/#!msg/gqrx/eqAcCAg0i84/8qNdJkJKBgAJ

and this bug report:
https://bugs.launchpad.net/ubuntu/+source/gqrx-sdr/+bug/1637774

Could you please try the AppImage and see if it works for you?

csete, 2016-11-21:
Any luck with the AppImage?

michelinok, 2016-11-23:
OMG...didn't see your answer :°(
I'll try again on another machine as soon as i can (i've reinstalled my machine with 16.04 due to an emergency in the gqrx usage).
I'll let you know if it works as soon as i get to my office (monday)

Sorry for the late response.

'73 de IU5HES

csete, 2016-11-23:
No problem. If you can use 16.04 it's fine :-)

michelinok, 2016-11-23:
yes...but you've wasted time answering... :°(

rhaamo, 2016-11-27:
Hi.

I may have the same problem.
I've got the segfault using the package then I tried to build from latest git master and same segfault problem.

I have tried the AppImage ```Gqrx-2.5.3.glibc2.17-x86_64.AppImage``` but I get:

```
/bin/bash: ../lib/x86_64-linux-gnu/libtinfo.so.5: no version information available (required by /bin/bash)
linux; GNU C++ version 4.8.4; Boost_105400; UHD_003.010.000.000-release

Gtk-Message: Failed to load module "atk-bridge"
/tmp/.mount_ENKL9X/usr/bin/gqrx-sdr: symbol lookup error: /usr/lib/x86_64-linux-gnu/libpangoft2-1.0.so.0: undefined symbol: hb_buffer_set_cluster_level
```

Any hints or things to try about the segfault issue ?

csete, 2016-11-27:
Hi @rhaamo 

There is a v2.6 AppImage: https://github.com/csete/gqrx/releases/tag/v2.6
Please try that.

I hope the main issue will be resolved in Ubuntu, see the linked bug report.

As I wrote on the mailing list, a workaround could be to build gr-osmosdr without UHD support then rebuild gqrx with the new gr-osmosdr

rhaamo, 2016-11-27:
No luck with the AppImage, same error.
I've rebuilt gr-osmosdr package without UHD and it's OK.

Thanks !

csete, 2016-11-27:
Thanks for trying :-)

I didn't have time to try it myself, so I'm glad to know that this workaround does work.

dennisheitmann, 2017-01-09:
Does it mean that gr-osmosdr is broken or just incompatible with the current version of gqrx? Which steps at Ubuntu (or anywhere else) are necessary to repair the situation? For which package I have to e.g. file a bug report?
Many thanks in advance.

csete, 2017-01-09:
@dennisheitmann please see my comment from Nov 2016: https://github.com/csete/gqrx/issues/452#issuecomment-261347506 

The problem occurs in the UHD library, which is the driver for Ettus USRP devices. The problem will affect anything that links to this library, including gr-osmosdr and gqrx. Ettus has been trying to upload new packages to their PPA so that we can use them until Ubuntu is updated (I think there is a link in the email thread) but they have been having problems with the PPA.

m1geo, 2017-01-10:
Same as @rhaamo, No luck with `Gqrx-2.6-1.glibc2.17-x86_64.AppImage`.

> george@watt:~/Desktop$ ./Gqrx-2.6-1.glibc2.17-x86_64.AppImage 
> linux; GNU C++ version 4.8.4; Boost_105400; UHD_003.010.001.000-release
> 
> ./Gqrx-2.6-1.glibc2.17-x86_64.AppImage: symbol lookup error: /usr/lib/x86_64-linux-gnu/libpangoft2-1.0.so.0: undefined symbol: hb_buffer_set_cluster_level
> george@watt:~/Desktop$ 

Ah well, that idea can wait for a while then! Thanks for your hard work @csete de M1GEO

lnx64, 2017-01-14:
Hopefully this will get fixed soon.  I'm a first responder and use this software to listen to different frequencies across the spectrum quickly, and I'm stuck between a rock and a hard place.  I need 16.10 to fix a Chrome issue with Intel GPU's (the drivers are not being backported to fix it), and I need gqrx to work, so I have to stay with 16.04 as it has priority right now.

I understand it's not gqrx's fault really, but the other devs need to hurry up, 17.04 will be out soon.

csete, 2017-01-31:
Hi guys, good news!

The various PPAs are now updated to provide packages for Ubuntu 16.10, so the PPA instructions should also work for 16.10: http://gqrx.dk/download/install-ubuntu

Please make sure to have all GNU Radio and gqrx packages uninstalled before using the PPA.

I can't test it, so I would appreciate feedback.

derpeter, 2017-01-31:
I just updates all the repositorys, uninstalled the old version and used the new one from the repos.
I tryed the app image, the bin from the repo and also build current git master. Still segfaulting.

./gqrx 
linux; GNU C++ version 6.2.0 20161005; Boost_106100; UHD_003.010.001.001-release

gr-osmosdr v0.1.x-xxx-xunknown (0.1.5git) gnuradio 3.7.10
built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy soapy redpitaya 
Resampling audio 96000 -> 48000
BookmarksFile is /home/derpeter/.config/gqrx/bookmarks.csv
Segmentation fault (core dumped)

the bottom of an strace on this shows:

epoll_ctl(32, EPOLL_CTL_ADD, 31, {EPOLLIN|EPOLLERR|EPOLLET, {u32=3833772328, u64=93947653437736}}) = 0
write(31, "\1\0\0\0\0\0\0\0", 8)        = 8
epoll_ctl(32, EPOLL_CTL_ADD, 33, {EPOLLIN|EPOLLERR, {u32=3833772340, u64=93947653437748}}) = 0
socket(AF_INET, SOCK_DGRAM, IPPROTO_UDP) = 34
epoll_ctl(32, EPOLL_CTL_ADD, 34, {EPOLLIN|EPOLLPRI|EPOLLERR|EPOLLHUP|EPOLLET, {u32=3833772432, u64=93947653437840}}) = 0
setsockopt(34, SOL_SOCKET, SO_BROADCAST, [1], 4) = 0
sendmsg(34, {msg_name(16)={sa_family=AF_INET, sin_port=htons(49152), sin_addr=inet_addr("XX.XX.XX.255")}, msg_iov(1)=[{"\0\0\0\f\0\0\0a\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0"..., 36}], msg_controllen=0, msg_flags=0}, MSG_NOSIGNAL) = 36
select(35, [34], NULL, NULL, {0, 100000}) = 0 (Timeout)
close(34)                               = 0
--- SIGSEGV {si_signo=SIGSEGV, si_code=SEGV_MAPERR, si_addr=0x1} ---
+++ killed by SIGSEGV (core dumped) +++
Segmentation fault (core dumped)

not sure if this is same error or a new / different one.

csete, 2017-01-31:
@derpeter What I need feedback about is the packages in the PPA installed on a clean system, i.e. no other installations of gqrx or gnuradio. So please no appimage or source builds.

derpeter, 2017-01-31:
@csete Sorry by "the repositorys" i mean the PPA gqrx-sdr  2.6-gqrx2~yakkety  i just wanted to point out that the same happens with source builds. If i can provide any further information let me know.

csete, 2017-01-31:
Ok, thanks for the info. The crash originally occurred in UHD and perhaps it has not been fixed with the recent Ettus PPA package :(

csete, 2017-01-31:
I have rebuilt gr-osmosdr without UHD support so if you refresh the packages and install the updates it should be working now.

romeojulietthotel, 2017-01-31:
Does this problem relate to cxx11 issues?  I have used this recently for some apps to get around issues 

`-D_GLIBCXX_USE_CXX11_ABI=0`

Have a look at #https://github.com/EttusResearch/uhd/issues/87

csete, 2017-01-31:
The problem here occurs at runtime after compilation.

csete, 2017-02-05:
Guys,

Did anybody try the latest packages from the PPA?

csete, 2017-02-06:
@m1geo I'm confused by how you get all these errors. Perhaps you are trying to use the "ppa:gqrx/gqrx-sdr" alone? You must follow the instructions here:

http://gqrx.dk/download/install-ubuntu

and add all PPAs. Sorry, but you either take it all or nothing.

m1geo, 2017-02-06:
I use them all, exactly as on the page you link. However, I get this error:

> The following packages have unmet dependencies.
> libgnuradio-fcdproplus3.7.10 : Breaks: libgnuradio-fcdproplus0 but 10.0.1.0-myriadrf5~yakkety is to be installed

Don't blame me, I'm just the messenger!

I may try again on a virtual machine shortly.

csete, 2017-02-06:
libgnuradio-fcdproplus0 is the library that comes from the official Ubuntu. You have to uninstall it before you are allowed to install the one from the PPA. Should have been taken care of by the "sudo apt-get purge --auto-remove libgnuradio*"  but I gues it doesn't work.

m1geo, 2017-02-06:
Ok, let me poke around some more then. I'll purge everything and check that it goes. There's nothing "fancy" or self compiled; it came from the PPA or it came from Ubuntu default repos.

QRX while I try something.

m1geo, 2017-02-07:
Yes, it works! I had to manually purge `gnuradio` and `gnuradio-dev` for some reason, but it's working now!

[image]

csete, 2017-02-07:
Great, thanks for the update!

csete, 2017-04-29:
Updated the title since the problem remains in Ubuntu 17.04:
https://bugs.launchpad.net/ubuntu/+source/gqrx-sdr/+bug/1683151
https://bugs.launchpad.net/ubuntu/+source/gqrx-sdr/+bug/1687176

lnx64, 2017-05-15:
I have not had any of these problems and I'm now on 17.04.  Is it due to the fact I installed it originally on 16.04?

csete, 2017-05-15:
Hello @lnx64 

Are you using the Gqrx PPA instructions? If yes, then the gr-osmosdr library is built without UHD support and the problem does not exist. However, the packages in the Ubuntu repository are built with UHD support and this is what crashes.

lnx64, 2017-05-26:
I think I was using the PPA, so that explains it then.

LeoTindall, 2017-11-12:
I'm experiencing this issue despite installing via PPA on Ubuntu 17.04.

csete, 2017-11-12:
@LeoTindall, It looks like I forgot to disable UHD support when packaging. It should be fixed now. Please try with the updated packages (0.1.4.98-myriadrf11) and let me know.

LeoTindall, 2017-11-12:
Solved! Thanks so much.

10011090, 2018-04-15:
Hi!
I'm not able to run gnuradio on linux.
      # **ImportError**
## _**Cannot import gnuradio.

## the model path environment variable set correctly?
    ## All OS: PYTHONPATH

## Is the library path environment variable set correctly?
    ## Linux: LD_LIBRARY_PATH
   ##  Windows: PATH
    ## MacOSX: DYLD_LIBRARY_PATH

## (No module named _runtime_swig**_)

Any idea?

10011090, 2018-04-15:
Hi!
I'm not able to run gnuradio on linux.

# ImportError
## Cannot import gnuradio.

Is the model path environment variable set correctly?
    All OS: PYTHONPATH

Is the library path environment variable set correctly?
    Linux: LD_LIBRARY_PATH
    Windows: PATH
    MacOSX: DYLD_LIBRARY_PATH

(No module named _runtime_sw

Any idea?

argilo, 2020-10-18:
Closing, since the issue appears to have been resolved, and Ubuntu 16.10 and 17.04 have been unsupported for a long time.

closed 2020-10-18 by argilo