gqrx-sdr/gqrx #454: ABI compatibility mismatch with UHD library

stephensworld, 2016-11-21:
Hi,

I have a problem when opening gqrx application.

[image]

Maybe someone could help me.

Thanks
Stephan

csete, 2016-11-21:
You probably have a mixture of packages installed from source and from binary repositories, which almost always leads to conflicts at runtime. I also note that you run gqrx as root; please don't do that!

csete, 2016-11-21:
It could also be that you have everything built from source, but you have upgraded UHD after you have built gnuradio and so the versions mismatch.

simontudo, 2016-11-30:
Same error here. I installed everything following [this guide](http://gqrx.dk/download/install-ubuntu) on a clean Ubuntu 16.04 installation. Nothing built from source, everything from PPA.

csete, 2016-11-30:
It looks like the UHD packages have been upgraded from 3.10.0.0 to 3.10.1.0 and this breaks binary compatibility with our packages. We have to rebuild the gnuradio packages, so stay tuned.

csete, 2016-12-03:
Hi guys,

There should be a new build of gnuradio available: 3.7.10.1-1~ubuntu16.04.1myriadrf2
It should appear as an update in your package manager.

I'm not sure if we also have to rebuild gr-osmosdr and I'm away from my dev computers, so can you please try it and see if it resolves the error?

Thanks!

simontudo, 2016-12-05:
Hey, 
I just updated the sources and installed the upgrades and everything works fine! Thank you very much!

csete, 2016-12-05:
Sounds great - thanks for reporting back!

closed 2017-02-19 by csete