gqrx-sdr/gqrx #456: GQRX build from source

nic076, 2016-12-04:
Please tell me how to fix error?
nikolay@HackRF:~/gqrx.git/build$ cmake ..
-- Boost version: 1.58.0
-- Found the following Boost libraries:
--   system
--   program_options
Checking for GNU Radio Module: RUNTIME
 * INCLUDES=/usr/local/include
 * LIBS=/usr/local/lib/libgnuradio-runtime.so;/usr/local/lib/libgnuradio-pmt.so
GNURADIO_RUNTIME_FOUND = TRUE
Checking for GNU Radio Module: ANALOG
 * INCLUDES=/usr/local/include
 * LIBS=/usr/local/lib/libgnuradio-analog.so;/usr/local/lib/libgnuradio-runtime.so;/usr/local/lib/libgnuradio-pmt.so;/usr/local/lib/libvolk.so
GNURADIO_ANALOG_FOUND = TRUE
Checking for GNU Radio Module: AUDIO
 * INCLUDES=/usr/local/include
 * LIBS=/usr/local/lib/libgnuradio-audio.so;/usr/local/lib/libgnuradio-runtime.so;/usr/local/lib/libgnuradio-pmt.so
GNURADIO_AUDIO_FOUND = TRUE
Checking for GNU Radio Module: BLOCKS
 * INCLUDES=/usr/local/include
 * LIBS=/usr/local/lib/libgnuradio-blocks.so;/usr/local/lib/libgnuradio-runtime.so;/usr/local/lib/libgnuradio-pmt.so
GNURADIO_BLOCKS_FOUND = TRUE
Checking for GNU Radio Module: DIGITAL
 * INCLUDES=/usr/local/include
 * LIBS=/usr/local/lib/libgnuradio-digital.so;/usr/local/lib/libgnuradio-runtime.so;/usr/local/lib/libgnuradio-pmt.so
GNURADIO_DIGITAL_FOUND = TRUE
Checking for GNU Radio Module: FFT
 * INCLUDES=/usr/local/include
 * LIBS=/usr/local/lib/libgnuradio-fft.so;/usr/local/lib/libgnuradio-runtime.so;/usr/local/lib/libgnuradio-pmt.so
GNURADIO_FFT_FOUND = TRUE
Checking for GNU Radio Module: FILTER
 * INCLUDES=/usr/local/include
 * LIBS=/usr/local/lib/libgnuradio-filter.so;/usr/local/lib/libgnuradio-fft.so;/usr/local/lib/libgnuradio-runtime.so;/usr/local/lib/libgnuradio-pmt.so
GNURADIO_FILTER_FOUND = TRUE
Checking for GNU Radio Module: PMT
 * INCLUDES=/usr/local/include
 * LIBS=/usr/local/lib/libgnuradio-runtime.so;/usr/local/lib/libgnuradio-pmt.so
GNURADIO_PMT_FOUND = TRUE
CMake Error at CMakeLists.txt:117 (find_package):
  By not providing "FindPulseAudio.cmake" in CMAKE_MODULE_PATH this project
  has asked CMake to find a package configuration file provided by
  "PulseAudio", but CMake did not find one.

  Could not find a package configuration file provided by "PulseAudio" with
  any of the following names:

    PulseAudioConfig.cmake
    pulseaudio-config.cmake

  Add the installation prefix of "PulseAudio" to CMAKE_PREFIX_PATH or set
  "PulseAudio_DIR" to a directory containing one of the above files.  If
  "PulseAudio" provides a separate development package or SDK, be sure it has
  been installed.

-- Configuring incomplete, errors occurred!
See also "/home/nikolay/gqrx.git/build/CMakeFiles/CMakeOutput.log".

dranch, 2016-12-04:
What HW and OS is this on?  I assume you installed all the required dependencies per https://github.com/csete/gqrx#installation-from-source

--David

csete, 2016-12-04:
You have to install the Pulseaudio development package (libpulse-dev) or choose a different audio system.

I'm closing the issue as it is neither a bug nor a feature request. User support, including help with compiling gqrx from source, is provided through the mailing, see http://gqrx.dk/user-support

closed 2016-12-04 by csete

nic076, 2016-12-04:
Thanks for the answer. 
What HW and OS is this on?  Memory 2Gb, Processor Intel Pentium G2120, Ubuntu 16.04LTS, 64- bit, HackRF.
Installing (libpulse-dev) solved the problem Thank you!     :)