gqrx-sdr/gqrx #457: Feature request for Ettus (ext Ref)

omegatee, 2016-12-08:
Hi, Alex.

I miss the ability to select the external reference on Ettus.

This must be easy to implement, like in "antenna selection"

csete, 2016-12-08:
Hi Luis,

I think this request is already on the list, though I can't find the reference, but until implemented you might be able to select the reference through the device string, see:

http://sdr.osmocom.org/trac/wiki/GrOsmoSDR#UHDSourceSink

pwr-krn, 2019-01-06:
Hi Alexandru.

can you please implement it?
It is not working through the device string!

csete, 2019-01-08:
If it is not working through the device string it will most likely not work otherwise either. This bug is about adding a GUI widget for convenience, not about debugging the interface. I will leave the debugging part to those who actually have a USRP.

pwr-krn, 2019-01-08:
I wonder why it is not working with the device string, with USRP extio selecting external refernce and pps is working fine.

pwr-krn, 2019-01-12:
It turns out that the problem is with gr-osmosdr.
please see here:

http://lists.ettus.com/pipermail/usrp-users_lists.ettus.com/2019-January/058806.html