gqrx-sdr/gqrx #458: Limit Request

omegatee, 2016-12-08:
Hi, Alex.

Why is LNB LO on "Input Controls" limited to 7.900 MHz ?

A typical use for SDR is as IF for TV-LNB, whose LO is 9.750 or 10.600 MHz

csete, 2016-12-08:
Not sure I agree with you on the typical use, but in any case this is on the TODO list, see #343 

(it's not a trivial fix because it is the main frequency widget that has limitations)

omegatee, 2016-12-08:
Sorry to repeat #343. 

Taking apart the typical use or not (*), I don't understand those main frequency widget limitations.
Is it something more than simply adding a figure to the main frequency display?
I wouldn't ask you for this simple fix if it were too complex to implement. Is better to invest time in big problems than in cosmetic mods.

(*) among endless other uses, I usually monitor my local 10 GHz beacon with GQRX, and looks than ewpereira also works that band as stated in #343.

Thank you, Alex.

Luis

closed 2016-12-08 by omegatee

csete, 2016-12-08:
@omegatee, please rest assured that I consider this as a serious and very annoying functional limitation, not just a cosmetic problem. So issue #343 is currently tagged to be fixed for the next release.

Also, you shouldn't worry whether something is simple or complex to implement. If you have wish for a feature just post it and we take it from there. Even if I don't have time or interest in implementing something, somebody else may see it and work on it, then send me a patch. Luckily, gqrx is increasingly a team effort and I can also spend time on other projects :smile: 

I remember that some time ago I looked into it to see if there was a simple trivial fix but there wasn't, so that's why it is still an open issue to be addressed.