gqrx-sdr/gqrx #459: APT deviation 25Khz instead of 17Khz

sv1, 2016-12-12:
Since upgrading to version 2.6 I can no longer decode NOAA satellites. The deviation of 25Khz is too wide and inserts a lot of noise.

csete, 2016-12-14:
I realize there are some issues wrt. APT reception; however, I'm surprised that changing the deviation alone would cause such issues for you. I heard of people using "wide fm" mode with much larger deviation for receiving NOAA APT.

csete, 2017-01-06:
Hi,

I have re-added 17 kHz so that one can choose between 17 and 25 kHz. Will be in the next release.

sv1, 2017-01-06:
Hello Alex,

thank you very much for your work on this.

After making some more recordings of NOAA passes and trying to decode them it seems that my problem is not with the max deviation but with the recording audio level when setting Tau to off.

The decoded images do not have any noise but are saturated with white color.

I have tested this also with audio udp streaming to other applications (fldigi, dsd, qsstv) when setting Tau to off then the audio input reported is more than 100% (102-108%)

Trying to lower the audio level of the recording before feeding it to wxtoimg or any other application I have to go to 10% of the original recording. (sox -v 0.1 .....) before they stop complaining.

The audio gain slider has no effect on recording/udp streaming audio level.

csete, 2017-01-06:
Indeed, I have noticed the saturated audio output too. I don't yet know what happened but I will investigate it ASAP.

csete, 2017-01-06:
I can see the output from the FM demodulator goes above the max range and it saturates the audio. As a workaround, you can use a non-zero de-emphasis value which will effectively reduce the level but it wont affect the APT signal much.

csete, 2017-01-08:
I think it works better now, although audio level can still go too high when there is no carrier. Maybe I should add a level adjustment for recording and streaming.

csete, 2017-08-19:
Unfortunately, I have to postpone finishing this until 2.8