gqrx-sdr/gqrx #464: Does not run on Mac OS X 10.9.5: Symbol not found: _OSAtomicIncrement32Barrier

probonopd, 2016-12-17:
Does not run on Mac OS X 10.9.5:

```
/Volumes/gqrx-2.6-1/Gqrx.app/Contents/MacOS/gqrx 
gr-osmosdr 5e71de7f (0.1.5git) gnuradio 3.7.10.1
built-in source types: file fcd rtl rtl_tcp hackrf rfspace airspy soapy redpitaya 
INFO: Audio sink arch: osx
INFO: Using output audio device 'Built-in Output'.
BookmarksFile is /Users/telekom/.config/gqrx/bookmarks.csv
Number of audio input devices: 1
Number of audio output devices: 1
dyld: lazy symbol binding failed: Symbol not found: _OSAtomicIncrement32Barrier
  Referenced from: /Volumes/gqrx-2.6-1/Gqrx.app/Contents/Frameworks/../Frameworks/libusb-1.0.0.dylib
  Expected in: /usr/lib/libSystem.B.dylib

dyld: Symbol not found: _OSAtomicIncrement32Barrier
  Referenced from: /Volumes/gqrx-2.6-1/Gqrx.app/Contents/Frameworks/../Frameworks/libusb-1.0.0.dylib
  Expected in: /usr/lib/libSystem.B.dylib

Trace/BPT trap: 5
```

I do not want to update beyond Mac OS X 10.9.x due to a number of reasons.

csete, 2016-12-17:
Sorry, but as it has been discussed many times before, I can only build and test on OS X 10.11 - even on that version I barely know what I'm doing. I'm just not a mac developer.

Does Qt 5 work on OS X 10.9 at all? If yes, you can try building it yourself or using macports or homebrew.

csete, 2016-12-17:
By the way, if somebody finds out why my builds aren't working and if there is a simple fix, like a build option, I'd be happy try and rebuild. I'm building through Qt SDK, which in turn uses X Code.

closed 2017-08-26 by csete