gqrx-sdr/gqrx #467: Problems on Mac OS X 10.12

csete, 2016-12-18:
https://groups.google.com/d/msg/gqrx/ynYZTObaeqw/ocOz3L0XDwAJ

---

First part of error report:

Process:               gqrx [578]
Path:                  /Volumes/VOLUME/Gqrx.app/Contents/MacOS/gqrx
Identifier:            dk.gqrx.www
Version:               2.6 (0)
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           gqrx [578]
User ID:               501

Date/Time:             2016-12-18 15:39:40.340 +0000
OS Version:            Mac OS X 10.12 (16A323)
Report Version:        12
Anonymous UUID:        E11B4E12-4012-5DF3-6BAA-00EDEAF18B7D

Time Awake Since Boot: 180 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Application Specific Information:
abort() called
terminating with uncaught exception of type boost::exception_detail::clone_impl<boost::exception_detail::error_info_injector<boost::system::system_error> >: resolve: Host not found (authoritative)

Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   libsystem_kernel.dylib        	0x00007fff9cdb6dda __pthread_kill + 10
1   libsystem_pthread.dylib       	0x00007fff9cea1797 pthread_kill + 90
2   libsystem_c.dylib             	0x00007fff9cd1c440 abort + 129
3   libc++abi.dylib               	0x00007fff9b87e85a abort_message + 266
4   libc++abi.dylib               	0x00007fff9b8a3c37 default_terminate_handler() + 243
5   libobjc.A.dylib               	0x00007fff9c3acf33 _objc_terminate() + 124
6   libc++abi.dylib               	0x00007fff9b8a0d69 std::__terminate(void (*)()) + 8
7   libc++abi.dylib               	0x00007fff9b8a07de __cxa_throw + 121
8   libgnuradio-blocks.3.7.10.1.dylib	0x000000010c1e9e11 void boost::throw_exception<boost::system::system_error>(boost::system::system_error const&) + 257
9   libgnuradio-blocks.3.7.10.1.dylib	0x000000010c1e9ccd boost::asio::detail::do_throw_error(boost::system::error_code const&, char const*) + 93
10  libgnuradio-blocks.3.7.10.1.dylib	0x000000010c1e7614 boost::asio::ip::basic_resolver<boost::asio::ip::udp, boost::asio::ip::resolver_service<boost::asio::ip::udp> >::resolve(boost::asio::ip::basic_resolver_query<boost::asio::ip::udp> const&) + 100
11  libgnuradio-blocks.3.7.10.1.dylib	0x000000010c20bec2 gr::blocks::udp_sink_impl::connect(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&, int) + 786
12  libgnuradio-blocks.3.7.10.1.dylib	0x000000010c20b72d gr::blocks::udp_sink_impl::udp_sink_impl(unsigned long, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&, int, int, bool) + 461
13  libgnuradio-blocks.3.7.10.1.dylib	0x000000010c20b44b gr::blocks::udp_sink::make(unsigned long, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&, int, int, bool) + 75
14  dk.gqrx.www                   	0x000000010b93648c udp_sink_f::udp_sink_f() + 556
15  dk.gqrx.www                   	0x000000010b93615f make_udp_sink_f() + 31
16  dk.gqrx.www                   	0x000000010b87befa receiver::receiver(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, unsigned int) + 3418
17  dk.gqrx.www                   	0x000000010b869817 MainWindow::MainWindow(QString, bool, QWidget*) + 1175
18  dk.gqrx.www                   	0x000000010b86446b main + 5147
19  libdyld.dylib                 	0x00007fff9cc88255 start + 1

Thread 1:
0   libsystem_kernel.dylib        	0x00007fff9cdb74e6 __workq_kernreturn + 10
1   libsystem_pthread.dylib       	0x00007fff9ce9e7c5 _pthread_wqthread + 1426
2   libsystem_pthread.dylib       	0x00007fff9ce9e221 start_wqthread + 13

Thread 2:
0   libsystem_pthread.dylib       	0x00007fff9ce9e214 start_wqthread + 0
1   ???                           	0x0006000700040006 0 + 1688879925297158

Thread 3:
0   libsystem_pthread.dylib       	0x00007fff9ce9e214 start_wqthread + 0
1   ???                           	0x0000700004457b90 0 + 123145373973392

Thread 4:
0   libsystem_kernel.dylib        	0x00007fff9cdb74e6 __workq_kernreturn + 10
1   libsystem_pthread.dylib       	0x00007fff9ce9e7c5 _pthread_wqthread + 1426
2   libsystem_pthread.dylib       	0x00007fff9ce9e221 start_wqthread + 13

Thread 5:
0   libsystem_pthread.dylib       	0x00007fff9ce9e214 start_wqthread + 0
1   ???                           	0x000070000455db80 0 + 123145375046528

Thread 0 crashed with X86 Thread State (64-bit):
  rax: 0x0000000000000000  rbx: 0x0000000000000006  rcx: 0x00007fff543a1578  rdx: 0x0000000000000000
  rdi: 0x000000000000150b  rsi: 0x0000000000000006  rbp: 0x00007fff543a15a0  rsp: 0x00007fff543a1578
   r8: 0x00007fff9cd45670   r9: 0x00007fff9cd07a70  r10: 0x0000000008000000  r11: 0x0000000000000206
  r12: 0x00007fff543a1700  r13: 0x0000000000000030  r14: 0x00007fffa597e3c0  r15: 0x0000000000000008
  rip: 0x00007fff9cdb6dda  rfl: 0x0000000000000206  cr2: 0x00007fffa5960128

mehdideveloper, 2017-02-28:
I don't have this problem.
OS X 10.12.2
Gqrx v2.6-73-g80f008f
Boost 1.63

And I haven't commented out the specific line in /etc/host.

csete, 2017-02-28:
Hi @mehdideveloper 

Thanks for the info! Can you tell me which version of Qt you are using. Thanks.

mehdideveloper, 2017-02-28:
Hi @csete 
Qt 5.6.1

erikarn, 2017-05-07:
Hi,

FreeBSD has a similar issue. rtl_tcp binds to 127.0.0.1 ; gqrx tries 'localhost' and just keeps looping over on the IPv6 localhost address, getting ECONNREFUSED each time.

Is it possible to make gqrx actually break out of the connect loop and just throw up a UI failure showing the IP address and port that it failed on?

erikarn, 2017-05-07:
In fact my problem is it gets stuck in a connect() loop. I'll open up another ticket.

thomasrussellmurphy, 2020-03-29:
I encountered this with the latest MacPorts version and fixed it with the update to `/etc/hosts`, ensuring an inclusion of a `localhost` line. The QT5 error message indicating to run from a console was not helpful as no other error message was displayed on the console upon program exit.