gqrx-sdr/gqrx #469: Partial waterfall redrawing

Cat-Ion, 2016-12-21:
Sub-issue of #393: When scrolling the waterfall along the frequency axis, it's possible (and preferable) to draw only the newly visible sections of it instead of the complete waterfall.

Cat-Ion, 2017-01-07:
An implementation of this is part of #461

csete, 2017-01-07:
So, this issue, which is part of issue #393 has its implementation as part of pull req #461 ... Sorry, but this process is too confusing and is not going to work out for me. I suspect that others who may come by will also find this confusing and just move on.

I'm really missing a clear and concise problem definition, i.e. what problem are you trying to solve? The original issue #393 had one sentence that could be considered as such, but already the second sentence starts talking about design and implementation details. 

This also applies to your comment from few weeks ago https://github.com/csete/gqrx/pull/461#issuecomment-268928468 where you give an excellent description of the changes you have made (should have been in the original text of the PR) but still no description of what problem you are trying to solve.

Without a clear problem definition I will not be able to spend any time on this. You have to understand that you are in a very tough competition. There are already 100+ open issues, many of which are critical and annoying bugs, so I can not justify spending time on reviewing code that I don't even understand what it's supposed to do.

So, I would recommend the following.

First, please make sure you understand the difference between a problem definition and design/implementation. First you define a problem, then you design and implement a solution. A good problem definition is very important for a good design and implementation in particular if others have to evaluate the solution.

Then close issues #393, #468 and #469. Create a new issue where you describe the problem. Don't talk about design and implementation details or issues; leave them to the pull request.

You can close pull request #461 and create a new one, or just continue with #461 if you wish, but please update/replace the original description with the overview you provided in https://github.com/csete/gqrx/pull/461#issuecomment-268928468 or something similar.

What do you think?

Cat-Ion, 2017-01-07:
I can close the issues, but to be honest I'm getting rather frustrated. If you don't want me to talk about the implementation, it's going to have a very short description (keep old stuff around so we can zoom etc). First you wanted me to separate the issue into multiple parts, even though I was just talking about implementation details and asking for an opinion on them, now you're unhappy that I'm working on the issues I've opened in one branch, where IMHO opening several new branches and pull requests wouldn't make anything clearer. The problem I'm trying to solve is still "You can't zoom around the waterfall and we should fix that", wether I ask about advice on implementation details in the issue or not.

If all you want changed is to have a new issue with a minimal description and an updated overview for #461 I can do that, but  I am (and please don't see this as a threat) getting rather close to just keeping this in my branch for my own use, or for whoever else feels like merging it.

csete, 2017-01-08:
All I am trying to do is guide you in a direction where your work has a chance of being considered instead of just being ignored. I'm quite disappointed that you can't see that because I have spent quite some time trying to explain myself.

> If you don't want me to talk about the implementation, it's going to have a very short description

I didn't say that. What I said was that I want to see a clear and concise problem definition before I can form an opinion about implementation details.

> First you wanted me to separate the issue into multiple parts, even though I was just talking about implementation details and asking for an opinion on them, now you're unhappy that I'm working on [...]

I have acknowledged that it was a mistake to open multiple issues and it was a clear indication that I didn't understand what you are trying to do: https://github.com/csete/gqrx/issues/393#issuecomment-268922261

Anyway, the fact remains that this topic is a mess in its present state and it should be cleaned up if you want people to give you feedback. I proposed a way forward that IMHO would present your work in a clearer way and I'm sorry that you can't see that.

> The problem I'm trying to solve is still "You can't zoom around the waterfall and we should fix that"

We can zoom. It may not be the way you think we should be able to zoom and that is exactly why a good description of the problem from a functional perspective is important.

cfr34k, 2017-01-09:
Hello everybody,

I'd like to join in as I have a feeling the discussion about this issue (and the #461 pull request) is getting a little out of hand causing frustration on both sides. In the hope to get around this, I have some comments and suggestions.

I have done some testing on the patch proposed in #461 and have to say it makes zooming around the waterfall and FFT charts much more intuitive than it is in the current gqrx version. @csete, http://temp.tkolb.de/gqrxzoom.mp4 is a short video (1 Minute, 60 MB) showing the behaviour of the waterfall with the patch. Note that no data in the waterfall is lost during all the zooming and scrolling action and the history is always aligned correctly to the frequency axis.

The zoom-able waterfall is quite helpful when you want to zoom to a narrow signal that is only transmitted very sporadically, like in the low power devices band shown in the video. Without the patch, you would have to zoom in partially and wait for the signal to reappear to zoom in further. Also, it makes aligning the demodulator much easier for such signals. IMHO, considering its usefulness, it would be a shame if this work was not integrated.

Regarding the process I have a suggestion:

Maybe the sub-issues should not be tracked on this repository, as they are not related to already integrated code, but the pull request #461. Instead, I suggest to track them in @Cat-Ion's repository. @Cat-Ion could give regular updates about them in #461 or just finalize that in its current state and open new pull requests for future performance improvements.

Keep up the good work!

Greetings,
cfr34k

csete, 2017-01-10:
Thanks @cfr34k for the demo and info; it's nice to see how it works. I still wish though that we could have a few sentences describing how it supposed to work in the body of the feature request and pull request so that people who see this can see the objectives instead of having to dig that info out from comments and source code. I think this work is loosing valuable feedback because people don't know what it is about.

Anyway, I don't want to spend anymore time on discussing these things. I am already spending several hours every day reading and writing gqrx-related emails and I am quite fed up with it. Just do whatever you want to do in whichever way you want to do it and we will take it from there. I promise to take a look at it as time and interest permits it.

Flamewires, 2017-01-24:
I'll throw in that I checked this out too and like how the new zoom/scroll works. Makes everything more fluid, and upon zooming, you can still see the previous waterfall, which makes tuning easier.