gqrx-sdr/gqrx #472: Memory leak running gqrx on RPi3

g7uhn, 2017-01-03:
Have tried running gqrx for long periods on a RPi3 (Raspbian/ Pixel desktop) with the aim of running an automatic receiver for exibition/ wall-mount display.  After some hours the RPi3 system slows to a halt, checking system logs (attached below) indicated there is a memory leak in the gqrx process.

Attached image shows CPU and memory usage logged on a basic RPi3/ Pixel/ gqrx system over 24 hours.  The first (flat) period is the RPi3 running the OS and desktop only, then gqrx is started (DSP running) just before 07:00, memory usage climbs steadily.

Hardware = RPi3 + RTL-SDR USB dongle
OS = Raspbian Jessie 2016-11-25 with Pixel desktop (accessed by VNC, VNC disconnected during test)
gqrx = gqrx-2.6-rpi3-1 (small window size and minimised during test)
SDR config = 300k sample rate, 137.1 MHz, NFM, deviation APT, Tau off, FFT 10fps

[image]

[kern-log-1.txt](https://github.com/csete/gqrx/files/683463/kern-log-1.txt)

csete, 2017-01-05:
I did some testing myself, only for a few hours, and it seems to me that the issue only exists on the Raspberry Pi. This is consistent with what I expected because I always do long duration tests before a release, I just didn't do it on the Raspberry Pi.

g7uhn, 2017-01-06:
Hi Alex, thanks for looking into this.  Interestingly my RPi3 system continues to work despite all the memory being gobbled up for quite some time until something happens that causes the RPi3 to go non-responsive for a period of time before it kills gqrx...  I've had it running for the last couple of days and made some observations, see attached.  Might be useful to you and let me know if there's any more help I can provide.

Andy

[RPi automatic wxsat debug - 6 Jan 2017.pdf](https://github.com/csete/gqrx/files/691069/RPi.automatic.wxsat.debug.-.6.Jan.2017.pdf)

csete, 2017-01-07:
Thanks for the update.

When the RAM is consumed the OS will start using swap space if there is any allocated. But since disk I/O is much slower that RAM the OS will become sluggish and very slow to respond. The OS will the start killing apps when it also runs out of swap space.

Which tools are you using to log memory usage? There are various command line tools like "free" that will also show you the amount of swap used.

The next steps for me are to use a memory profiler that tells exactly which functions are leaking, but this has to be done on target and I have had problems running in on the RPI. So I am focusing on restoring the FM and filter settings first (issue #473) and to fix a serious problem with the FM demodulator (issue #459).

g7uhn, 2017-01-07:
I'm using SNMP and am running MRTG on another machine that runs 24/7 to
poll the RPi3 under test.  So I can monitor anything that has an entry in
the MIB.  I'll add "Available Swap Space" as I can see its OID here:
http://www.debianadmin.com/linux-snmp-oids-for-cpumemory-and-disk-statistics.html

Anything else you can think of worth adding to the monitoring?

PaulSlocum, 2017-08-03:
I have the same problem running on RPi3.  After running Gqrx for about 3 days it slows down and later freezes and becomes unresponsive.  We have six systems running with different settings, and they all slow and freeze up around the same time after 3 days.

csete, 2017-08-03:
Sorry for not working on this but I just couldn't prioritize it.

The good news is that Raspbian is being updated to Debian Stretch, which comes with a quite recent gqrx and gnuradio. We can then try using the gqrx package that comes with the OS and hopefully there will be no need for my binaries any more. It will also make debugging easier.

csete, 2017-11-12:
There is a new Gqrx release 2.9 for Raspbian Stretch. This release uses as much of the official packages as possible. I have only tested it for about an hour, but would be very interested to know if it still has the memory leak.

g7uhn, 2017-11-13:
Sounds cool, I'll rebuild a RPi3 with Stretch and give it a go this week,
thanks Alex.

Andy

g7uhn, 2017-11-18:
Took longer than expected to run this test, busy week, but it looks massively improved on memory usage, see attached.  I'll leave this running and see how long it lasts.

[image]

csete, 2017-11-18:
Looks promising Andy, thanks for testing!

g7uhn, 2017-11-18:
No worries, happy to do anything I can to help.  Thanks for all the work
and pointing out the update :-)

g7uhn, 2017-11-25:
Over a week later and GQRX 2.9 is still running on the RPi3.  Memory usage seems to have flattened out.  Looks really good.
[image] 
[image]

csete, 2017-11-25:
Andy,

This is very good news! :)
It's also cool to see that it could keep the 50% CPU load for so long time without any thermal shutdown or similar.

I guess we can conclude that the problem was in external library and got fixed. If you agree I will close this bug.

g7uhn, 2017-11-25:
Agree on closing the issue.  Good stuff!

Good point on temperature although this RPi3 is outside the house in my
cold, damp garage so the ambient is below room temperature.  Currently
showing CPU temperature = 71deg

csete, 2017-11-25:
Hopefully the new SDR will run with lower CPU load ;)

closed 2017-11-25 by csete