gqrx-sdr/gqrx #473: Feature request:  remember filter settings on startup

g7uhn, 2017-01-03:
A request for filter settings to be remembered on gqrx startup (ideally editable in config file).  Ideally all filter settings but my intended use is to recall settings for NOAA APT weather satellite reception e.g.

Demod = NFM
Deviation = APT (25 kHz)
Tau = off
Filter bandwidth ~ 38 kHz

Specifying these in the config file would allow me to launch gqrx for any of the given three satellites by choosing from three different config files (which correspond to the three different downlink frequencies).

Many thanks for all the hard work that goes into gqrx!

pinkavaj, 2022-11-07:
@g7uhn I have tested it and it seems it is implemented in latest version can you confirm?

vladisslav2011, 2022-11-07:
Current master does not save/restore the filter shape and AM/AM sync DCR setting and maybe some other settings...

argilo, 2022-11-07:
Everything mentioned in the original issue is now saved. It looks like the missing pieces were deviation (added in 5a9e04d432f0c9c34424fc0cb553749a4dbfcb7d) and de-emphasis (added in 52dd3daac558598d60e1d1e1640449bd17823504). So I think it makes sense to close this issue.

closed 2022-11-07 by argilo