gqrx-sdr/gqrx #478: Building gqrx on Raspberry PI

F5OEO, 2017-01-09:
Notes and requirements to properly build gqrx on Raspberry pi 2 or 3, Jessie based.

csete, 2017-01-09:
The problem is that Gqrx and GNU Radio needs Volk to run efficiently on the PI but the compiler that comes with Raspbian is not able to build volk, probably due to ABI compatibility with earlier PIs. Therefore, gqrx that comes with Raspbian runs very poorly.

The solution (for me) was to build everything (except Qt) from source, starting with the compiler. My detailed notes are below. Working directory is /home/pi/sdr/ and the runtime environment is /home/pi/sdr/runtime

---

Start with a fresh Raspbian Jessie (tested 2016-05-07). Configure and bring it up to date.

Install gcc into /home/pi/gcc/
NB: /home/pi/gcc/lib must be copied to the SDR runtime (TBD how much?)

Install:
cmake git libusb-1.0-0-dev libasound2-dev python-cheetah libudev-dev
time vim htop mc automake autoconf libtool

Create 1 GB swap file:
sudo dd if=/dev/zero of=/swapfile bs=1M count=1024
sudo mkswap /swapfile
sudo swapon /swapfile

export CC=/home/pi/gcc/bin/gcc
export CXX=/home/pi/gcc/bin/g++

FFTW:
sets it's own compiler flags, so don't add -funsafe-math-optimizations
wget http://www.fftw.org/fftw-3.3.5.tar.gz
./configure --prefix=/home/pi/sdr/runtime --enable-shared=yes --enable-single --enable-neon --disable-doc
make -j2

ORC:
wget https://gstreamer.freedesktop.org/src/orc/orc-0.4.26.tar.xz
./configure --prefix=/home/pi/sdr/runtime --enable-gtk-doc-html=no
make -j4

BOOST:
wget http://downloads.sourceforge.net/project/boost/boost/1.61.0/boost_1_61_0.tar.bz2
echo "using gcc : : /home/pi/gcc/bin/g++ ; " >> tools/build/src/user-config.jam
./bootstrap.sh --prefix=/home/pi/sdr/runtime
./b2 -j 2
./b2 install

VOLK:
wget http://libvolk.org/releases/volk-1.3.tar.gz
cmake .. -L -DCMAKE_INSTALL_PREFIX=/home/pi/sdr/runtime
(make test fails, why?)

GNURADIO:
wget http://gnuradio.org/releases/gnuradio/gnuradio-3.7.10.1.tar.gz

cmake .. -L \
    -DCMAKE_INSTALL_PREFIX=/home/pi/sdr/runtime             \\
    -DFFTW3F_INCLUDE_DIRS=/home/pi/sdr/runtime/include/     \\
    -DENABLE_INTERNAL_VOLK=OFF  \\
    -DENABLE_GR_FFT=ON          \\
    -DENABLE_GR_FILTER=ON       \\
    -DENABLE_GR_ANALOG=ON       \\
    -DENABLE_GR_DIGITAL=ON      \\
    -DENABLE_GR_AUDIO=ON        \\
    -DENABLE_GR_FCD=ON          \\
    -DENABLE_GR_NOAA=OFF        \\
    -DENABLE_GR_FEC=OFF         \\
    -DENABLE_GR_COMEDI=OFF      \\
    -DENABLE_GR_PAGER=OFF       \\
    -DENABLE_GR_QTGUI=OFF       \\
    -DENABLE_GR_TRELLIS=OFF     \\
    -DENABLE_GR_UTILS=OFF       \\
    -DENABLE_GR_VOCODER=OFF     \\
    -DENABLE_GR_WAVELET=OFF     \\
    -DENABLE_GR_WXGUI=OFF       \\
    -DENABLE_GR_CHANNELS=OFF    \\
    -DENABLE_GR_ZEROMQ=OFF

make -j3

RTLSDR:
git clone git://git.osmocom.org/rtl-sdr.git rtl-sdr.git
cmake .. -DCMAKE_BUILD_TYPE=release -DCMAKE_INSTALL_PREFIX=/home/pi/sdr/runtime -DDETACH_KERNEL_DRIVER=ON
sudo cp rtl-sdr.rules /etc/udev/rules.d/54-rtl-sdr.rules

HACKRF:
git clone https://github.com/mossmann/hackrf.git hackrf.git
cmake .. -DCMAKE_BUILD_TYPE=release -DCMAKE_INSTALL_PREFIX=/home/pi/sdr/runtime -DINSTALL_UDEV_RULES=OFF
sudo cp libhackrf/53-hackrf.rules /etc/udev/rules.d/

AIRSPY:
git clone https://github.com/airspy/host.git airspy-host.git
cmake .. -DCMAKE_BUILD_TYPE=release -DCMAKE_INSTALL_PREFIX=/home/pi/sdr/runtime
sudo cp airspy-tools/52-airspy.rules /etc/udev/rules.d/

GR-FCDPROPLUS:
git clone https://github.com/dl1ksv/gr-fcdproplus.git gr-fcdproplus.git
<< disable python >>
cmake .. -L -DCMAKE_BUILD_TYPE=release -DCMAKE_INSTALL_PREFIX=/home/pi/sdr/runtime

SOAPYSDR:
git clone https://github.com/pothosware/SoapySDR.git soapysdr.git
git checkout soapy-sdr-0.5.3
cmake .. -DCMAKE_BUILD_TYPE=release -DCMAKE_INSTALL_PREFIX=/home/pi/sdr/runtime

LIBOSMO-DSP
git clone git://git.osmocom.org/libosmo-dsp libosmo-dsp.git
git clone https://github.com/osmocom/libosmo-dsp.git libosmo-dsp.git
autoreconf -i
./configure --prefix=/home/pi/sdr/runtime

GR-IQBAL
git clone git://git.osmocom.org/gr-iqbal gr-iqbal.git
cmake .. -DCMAKE_BUILD_TYPE=release -DCMAKE_INSTALL_PREFIX=/home/pi/sdr/runtime

GR-OSMOSDR:
(git clone git://git.osmocom.org/gr-osmosdr gr-osmosdr.git)
git clone https://github.com/csete/gr-osmosdr-gqrx.git gr-osmosdr-gqrx.git
merge airspy and rfspace branches into build
cmake .. -DCMAKE_BUILD_TYPE=release -DCMAKE_INSTALL_PREFIX=/home/pi/sdr/runtime
make -j2

GQRX:
sudo apt-get install qt5-default qt5-qmake qtbase5-dev libqt5svg5-dev portaudio19-dev
git clone https://github.com/csete/gqrx.git gqrx.git
cmake .. -DCMAKE_BUILD_TYPE=release -DLINUX_AUDIO_BACKEND:STRING=Portaudio -DCUSTOM_AIRSPY_KERNELS=on -DCMAKE_INSTALL_PREFIX=/home/pi/sdr/runtime
make -j2

F5OEO, 2017-01-10:
Which gcc did you choose to use after having issues with the one in Raspbian?
 By using 4.9.2 I got 
/home/pi/sdr/volk-1.3/kernels/volk/asm/neon/volk_32f_x2_dot_prod_32f_neonasm_opts.s:46: Error: selected processor does not support ARM mode `sbfx r11,r1,#2,#1'

Did you have this kind of message with the orginal gcc ?

For now 
- boost is compiled (with some failed)
- FFTW
- ORC

csete, 2017-01-10:
Yes, that is exactly the problem with the GCC that comes with raspbian and was why I started with building GCC. I think it is because they want to keep compatibility with old raspberry pi.

I used the latest GCC available at that time, I think it was 6.2 but  Idon't think it matters much. I can try to send you my binaries if you want - it may save you some time. But it will be a very large (many GB) download.

csete, 2017-01-10:
PS regarding boost, fftw and ORC - do NOT compile anything with gcc 4.9 - use the same compiler for everything!

900fy, 2017-02-06:
hello, 

I also have been thinking about compiling gqrx for RPI
is it possible to cross compile GQRX code on x86 system instead of doing it inside of RPI?
I think it could be much faster and handy, have you ever tried to do that?

csete, 2017-02-06:
@900fy of course it is possible but it requires that you set up a build environment with all the packages required for building on the host system. You are more than welcome to try, in fact, I would really welcome if somebody else would take this task over as I already have tons of other things on my list preventing me from doing any active development on this project.

900fy, 2017-02-06:
I've already made some research in this topic and I hope to set up build environment during this week. If I succeed I will give you a feedback. 
I plan to make few modifications to user interface for better ergonomics on 7" touch screen on my RPI
btw, thank you for featuring my video on your website :) 
http://gqrx.dk/blog/pi-sdr-raspberry-pi-3-7-touchscreen-rtl-sdr-gqrx

csete, 2017-02-06:
Sounds great. The only critical thing to remember is not to use the official Raspbian compiler since it is not able to build optimized code. Keep us posted!

Indeed, the user interface needs modifications to work on small screens. The quickest thing would probably be to create a new dock widget that would contain the most essential controls and then only show this dock widget. Anyway, that's a different topic and feel free to create a new issue if you want to discuss it with people.

F5OEO, 2017-02-06:
Sorry didn't finish the job ...was at the step of Gcc compiling !
Instead of having a cross-compile environment, I used Qemu-static which is convenient as your are on Raspbian directly. If needed, I have few scripts to automate.

Great that @900fy follows ...

KH6JAG, 2017-05-15:
I'm going to try to install GQRX on an ASUS Tinker Board.  I'm trying to automate a amateur,  weather and Thumbsat satellite capability at our HAM station in the South Florida Science Center.  I have very limited knowledge of  Linux but I'm willing to give it a try.  If I give your instructions a try is this still an active effort so that if I have problems I can get help or ask questions?  -OR- 
Do have a better starting point you can point me to.  TNX

vk4tec, 2017-05-15:
I will try and help where I can

Andrew

csete, 2017-05-15:
@KH6JAG These are not intended as step-by-step instructions for installing anything. These are just my notes from building a raspberry pi optimized version of gnuradio and gqrx. I posted them so that others who want to experiment can try it.

If your objective is to run gqrx, then you should start with a platform that is known to work. This could be a PC running Ubuntu 16.04 LTS, a Raspberry Pi 3 or a Mac, see http://gqrx.dk/download

In any case, I prefer user support to happen through the mailing list and keep github issues for tracking bugs and feature request, see http://gqrx.dk/user-support

csete, 2017-05-15:
@F5OEO If you don't mind, I will close this issue as there is nothing to do for me

F5OEO, 2017-05-15:
No problem could close...

closed 2017-05-15 by csete