gqrx-sdr/gqrx #480: [Suggestion] Poor performance on RPi 3

hemee, 2017-01-18:
It's clear why GQRX performs poorly on RPi 3, but I'd like to make a little contribution by suggesting to build it with these flags (CFLAGS or CXXFLAGS whatever, just do `export`):
`-O2 -marm -mabi=aapcs-linux -march=armv8-a+crc -mfloat-abi=hard -mfpu=neon-fp-armv8 -funsafe-math-optimizations -mhard-float -mlittle-endian -mtune=cortex-a53 -mvectorize-with-neon-quad`
I have been testing these flags by myself, and it produces very nice A53-optimized code. I believe it may improve some performance bottlenecks such as Fourier calculations.
Hope it will become a little step to get non-laggy GQRX for third raspberries.

csete, 2017-01-18:
Hi @hemee 

Thanks for the suggestion; however, we need more details about your setup and environment. We already provide optimized binaries for RPI2 and 3 that have been tested to work quite well, see http://gqrx.dk/download/gqrx-sdr-for-the-raspberry-pi

So, could you please tell us which binary gives you the poor performance and under which circumstances? Thanks.

hemee, 2017-01-18:
Yea, I got one from this same link for RPi 3. My system is custom Raspbian (lite version, all packages were obtained and configured by myself), but I think it's pretty same as any other Raspbian like NOOBS or other else. I keep it up-to-date.
It's not even poor, but as not fine as it meant to be I guess, because I can't use RPi's GQRX after my laptop. I understand that performance is not comparable, but it really can be better. I use my RPi for many desktop tasks and it does it well.
So, I don't see any reasons not to try this optimizations. They appears to make Quake 3 much more... erm, responsive. Dump1080 catches more planes too (in aggressive mode).

csete, 2017-01-18:
The point I want to make is that those builds are already optimized with flags similar to the one you posted, but most of the performance boost comes from using a custom compiler and volk.

Did you run the setup script as described in our guide? 

I don't mind trying your flags but I am very busy at the moment and I would like to avoid spending time trying something that at the end will not help. I am also in the middle of moving right now, so it's a bit difficult for me to set up RPI right now.

So, can you please confirm that

1. You have tried our package following the exact instructions in the included readme from http://gqrx.dk/download/gqrx-sdr-for-the-raspberry-pi
2. Your flags give a performance better than the ones listed under "Device support" http://gqrx.dk/download/gqrx-sdr-for-the-raspberry-pi

Also, please provide some details about what kind of SDR hardware and what settings you are using.

hemee, 2017-01-18:
Both statements are true. I executed setup script and was following the instructions. I'm using RTL-SDR Blog dongle, it works flawlessly on x64 GQRX build.

csete, 2017-01-18:
Ok, thanks for the info. But as you can see on that page, rtlsdr should work fine up to 2.4 Msps with our binaries so something is not right with your setup. Perhaps you have other gnuradio packages installed?

Anyway, I will compare your flags with the ones I use but it will probably take some time.

KC7NOA, 2017-03-10:
I can set sample rate to 3000000 and Rx bandwidth 60Mhz -- Odroid-C2 / LimeSDR / Ubuntu 16.04 64bit

argilo, 2020-11-28:
I'll close this off since it's outdated at this point.

closed 2020-11-28 by argilo