gqrx-sdr/gqrx #481: why gqrx can not find hackrf?!

stackprogramer, 2017-01-26:
Hi,when i install gqrx from command line apt-get install gqrx-sdr on new distribution ,both on deian or ubuntu it can not find hackrf!
i installed all dependency from this link:
https://mborgerson.com/getting-started-with-the-hackrf-one-on-ubuntu-14-04

when i type hackrf_info it returns serial number and id,every thing is ok.
`Found HackRF board.
Board ID Number: 2 (HackRF One)
Firmware Version: ...
Part ID Number: ...
Serial Number: ...`

now i had a question, it is necessary for using hackrf we should built from source again??
thanks for attention 
best regards stackprogramer

csete, 2017-01-26:
I have not packaged gqrx for Debian or Ubuntu so I do not know the answer.

As I recall, the gqrx package in Ubuntu 14.04 and Debian is very old, so it may simply not have hackrf support. You should be able to check that through the dependency tree, i.e. see if installing gqrx-sdr also installs libhackrf.

If you install hackrf support manually through source build, then you will also have to build gqrx and gr-osmosdr from source.

For Ubuntu we provide the latest version through PPA:
http://gqrx.dk/download/install-ubuntu

stackprogramer, 2017-01-26:
@csete thanks for reply i built gqrx again from source, now it can find hackrf one, it works like charm. 
thanks very much 
best regards stackprogramer

csete, 2017-01-26:
Great :)

closed 2017-01-26 by csete

cleberalves40, 2021-09-16:
Same problem with FreeBSD, so it have to compile from the Ports tree (gs-osmosdr dependency, this is the real problem).
Works like a charm, after that.