gqrx-sdr/gqrx #482: Interface issues with Hi DPI display

voltagex, 2017-01-27:
Gnome's scaling factor is set to 2.

[image]

csete, 2017-01-27:
Hi,

I am aware of high DPI issues when interacting with the spectrum widget, but I presume you are referring to the widgets going off screen? I don't think that's a high DPI issue, you should see the same effect by reducing the window size.

Anyway, I tried improving that in version 2.6 but it is highly dependent on the desktop theme and the Qt version. In the future I will no longer use the dock widgets. Until then, you can work around by dragging the splitters and give the widgets more room.

voltagex, 2017-01-27:
Hi,
When I first started the application the text on some of the buttons was
cut off / unreadable. Some buttons are also really tiny (halve the size of
my screenshots for an accurate representation).

Apparently the latest versions of Qt 5 are also better for screens like
mine.

argilo, 2023-10-03:
A lot of work has been done on adding proper high-DPI support to Gqrx, beginning in version 2.14.3. I would guess that this particular problem is no longer present, so I'll close off this issue. If that's not the case, let me know.

closed 2023-10-03 by argilo