gqrx-sdr/gqrx #484: No audio output, crash on changing audio output device

vitalyga, 2017-02-01:
Hi there, I have an issue with audio on gqrx. When I running the app with audio output device "default", it starts OK but no sound. When I entering the "Configure I/O devices" menu and choosing some other device, the app crashes with error message as follows:

-------------------------
Qt has caught an exception thrown from an event handler. Throwing
exceptions from an event handler is not supported in Qt. You must
reimplement QApplication::notify() and catch all exceptions there.

terminate called after throwing an instance of 'std::invalid_argument'
  what():  cannot disconnect edge multiply_const_ff0:0->pa_sink0:0, not found
Aborted
----------------------------------

Same crash and error message when I running gqrx with -r option and selecting any device from the list.
The options are:
- Default
- Built-in Audio Analog Stereo
- Loopback Analog Stereo
- Built-in Audio Digital Stereo (HDMI 2)

The OS is Gentoo, kernel 4.4.26

csete, 2017-02-01:
When you start the program and there is no audio, it is because you have to select a demodulator that produces any audio. As for the errors with a different audio device, it could be sample rate mismatch (gqrx only works at 48 kHz)

vitalyga, 2017-02-01:
You're absolutely right, it's my fault, I indeed didn't choose the demodulator. When I choosing it, all works as supposed.
As for the errors, I didn't change the default 48kHz sample rate.

csete, 2017-02-01:
What I meant with the sample rate is that the other devices may require something else than 48 kHz; it can not be changed in gqrx.

closed 2017-02-01 by vitalyga

csete, 2017-02-10:
Hi @vitalyga 

You can try again with the latest code from git if you wish. Hopefully this patch from @alexf91 fixes your problem also.